ghdl/ghdl #183: Should GHDL get error/warning IDs for filtering?

Paebbels, 2016-11-02:
I tried to implement a [compile script](https://github.com/UVVM/UVVM_Utility_Library/issues/1?ts=2) for [UVVM_Utility_Library](https://github.com/UVVM/UVVM_Utility_Library?ts=2). This library uses shared variables without protected types. They this problem in QuestaSim by [disabling certain message IDs](https://github.com/UVVM/UVVM_Utility_Library/blob/master/uvvm_util/script/compile_src.do?ts=2#L87).

Should GHDL print out messages IDs for errors and warnings to allow easier filtering?

**Example:**
> WARNING: D:\...\pkg.vhd:63:19:warning : [ANA-005] type of a shared variable must be a protected type

**Possible prefixes:**
* PARSE - paring the input file
* ANA - analysis
* ELAB - elaboration
* RUN - runtime
* ...

tgingold, 2016-11-02:
On 02/11/16 12:18, Patrick Lehmann wrote:
> I tried to implement a compile script
> <https://github.com/UVVM/UVVM_Utility_Library/issues/1?ts=2> for
> UVVM_Utility_Library
> <https://github.com/UVVM/UVVM_Utility_Library?ts=2>. This library uses [...]

Most of the infrastructure is present to do that.  I don't like the use 
of numbers, I much prefer the use of identifiers.  What about something 
like:

-Wno-shared-protected

We just need to find good names.

Tristan.

Paebbels, 2016-11-02:
My idea was about a label in the message text itself, so no new arguments are required.

But yes the gcc way with enable/disable warnings (`-Wno-***`) is also.
Can you easily map `-Wno-identifier` to an internal Ada error/warning identifier?

tmeissner, 2016-11-02:
"This library uses shared variables without protected types." Then it should be compiled with VHDL 93 standard enabled, shouldn't it?

tgingold, 2016-11-02:
On 02/11/16 14:11, Patrick Lehmann wrote:
> My idea was about a label in the message text itself, so no new
> arguments are required.
>
> But yes the gcc way with enable/disable warnings (|-Wno-***|) is also.
> Can you easily map |-Wno-identifier| to an internal Ada error/warning
> identifier?

I don't see any particular issue here.

tgingold, 2017-08-27:
It is now possible to select which warning to enable/disable.

closed 2017-08-27 by tgingold

svenn71, 2017-11-02:
How do I disable those errors from type of shared variable?

tgingold, 2017-11-03:
> On 02 Nov 2017, at 14:08, svenn71 <notifications@github.com> wrote:
> 
> How do I disable those errors from type of shared variable?

Use -frelaxed-rules

Tristan.