ghdl/ghdl #188: Should GHDL have a Gitter room?

Paebbels, 2016-11-08:
I got some GHDL related questions via Gitter on my private Gitter channel. Gitter can be used by all GitHub users and I think twitter accounts are also allowed. Reading Gitter is without restrictions.

Should GHDL have a Gitter room?

VUnit uses such a online chat room with great success. All messages are stored for a later review like in a mailing list. I think it could help people with short questions. Bigger problems should of cause go to an issue (or the mailing list?)

A nice feature is that Gitter shows notifications from GitHub like new pushes, merges or new releases (with link and description text).

What do you think?

tgingold, 2016-11-08:
On 08/11/16 11:17, Patrick Lehmann wrote:
> I got some GHDL related questions via Gitter on my private Gitter
> channel. Gitter can be used by all GitHub users and I think twitter
> accounts are also allowed. Reading Gitter is without restrictions.
>
> Should GHDL have a Gitter room?

Why not.  Can you create it ?

Paebbels, 2016-11-08:
I'm no member of the ghdl repo neither have I admin rights. So I'm not allowed to create a repository for GHDL. I can only select repositories from my own repository list and from organisations, where I have admin rights. I could create a new stand-alone room, but the name ghdl seems to be reserved or protected.

So you need to create it, or I need more rights :).

**How to create a linked room?**

1. Click on Create new community
2. Click on the link to select a github repository:  
   [image]
3. Select the repository from the list:
   [image]

That should do the work.

You can then configure the room (e.g. set moderators, restrict access, etc.). It should somewhere on the right side ask, to connect Gitter with a repository, so it can then display Git notifications like issues or merges.

tgingold, 2016-11-09:
Done, using ghdl1.

closed 2016-11-09 by tgingold

Paebbels, 2016-11-09:
You might want to strip the automatically generated Gitter badge's URL. It contains several needless URL parameters by default :).