ghdl/ghdl #190: Object subtype is not locally static

Paebbels, 2016-11-09:
In continuation of "not static" messages: issue #10, issue #151, I might have found another one.

Here is the code:
```VHDL
library ieee;
use      ieee.std_logic_1164.all;

entity e is
end entity;

architecture a of e is
  signal foo : std_logic_vector(7 downto 0);
  signal bar : std_logic;
  
  procedure proc(constant addr : in std_logic_vector; signal output : out std_logic) is
    alias addr_alias : std_logic_vector(addr'length downto 0) is addr;  -- inherited non-staticness from addr?
  begin
    -- case addr(1 downto 0) is     -- not static, even if addr is of kind constant
    case addr_alias(1 downto 0) is  -- see local constant declarations
      when others =>  output <= 'X';
    end case;
  end procedure;
begin
end architecture;
```

I used this command line: `ghdl.exe -a --std=08 .\not_locally_static.vhdl`
The result is the same with `-frelaxed-rules` enabled.

The error message is:
> .\not_locally_static.vhdl:19:31: object subtype is not locally static

In my opinion, `addr` should be static enough for a case statement :).
[not_locally_static.txt](https://github.com/tgingold/ghdl/files/581793/not_locally_static.txt)

See the attached file for more details and code variations. Please not, the length mismatch is not yet discovered. This might happen in elaboration time, right?

tgingold, 2016-11-09:
That's strange as there is very similar code in std_logic_1164.
Need to investigate.

tgingold, 2016-12-10:
After a second look, this code is clearly not valid in vhdl-93.
It should be valid in vhdl-08.

addr is not locally static (bounds are not known), so it cannot be used as expression of a case statement.

tgingold, 2017-10-22:
Implemented.

closed 2017-10-22 by tgingold