ghdl/ghdl #191: 'others' choice not allowed for an aggregate in this context

Paebbels, 2016-11-09:
Sorry for writing so many issues tonight :).

I have this code snippet:
```VHDL
library ieee;
use      ieee.std_logic_1164.all;

entity e is
end entity;

architecture a of e is
  type t_rec is record
    addr : std_logic_vector;
  end record;
  
  function init_signals(bits : natural) return t_rec is
    variable result : t_rec(addr(bits - 1 downto 0));
  begin
    result.addr   := (others => '0');
    -- result.addr   := (result.addr'range => '0');  -- this works
    return result;
  end function;
begin
end architecture;
```

I used this command to analyze:
`ghdl.exe -a .\cant_use_others_here.vhdl`.

As expected the error message(s) is:
> .\cant_use_others_here.vhdl:9:16: element declaration of unconstrained array type "std_logic_vector" is not allowed
> .\cant_use_others_here.vhdl:15:22: 'others' choice not allowed for an aggregate in this context

So yes. the used feature is new to VHDL-2008. So let's enable it:
`ghdl.exe -a --std=08 .\cant_use_others_here.vhdl`

But one error message survives:
> .\cant_use_others_here.vhdl:15:22: 'others' choice not allowed for an aggregate in this context

If I change the `others` keyword to `prefix'range` it's accepted.
[cant_use_others_here.txt](https://github.com/tgingold/ghdl/files/581936/cant_use_others_here.txt)

See the attached file for more details and code variations.

diogratia, 2016-11-10:
This case is found in IEEE Std 1076-1993, 7.3.2.2 Array aggregates, para 3, d.:

>The subtype of an array aggregate that has an others choice must be determinable from the context. That is, an array aggregate with an others choice may only appear
>... 
>    d. As a value expression in an assignment statement, where the target is a declared object, and the subtype of the target is a constrained array subtype (or subelement of such a declared object)

Which tells us the others choice can't be used here.

Now for the bad news. This is also seen in IEEE Std 1076-2008 9.3.3.3 Array aggregates, para 7 e).

ghdl appears to be correct in complaining in both revisions of the standard that the **others** choice cannot be used in this context.

This should be viewed as an analysis time requirement. How do you generate the range of associations represented by the **others** choice at analysis time without an index constraint?

Paebbels, 2016-11-10:
OK, but what's the difference between `others` as single keyword and `'range`. It is just the time/phase when the set of zeros is created?

So `others` is a keyword and can be used immediately at analysis time and `'range` is an attribute an can be delayed until elaboration time?

diogratia, 2016-11-10:
There's a basic concept missing here, that structure of an aggregate is defined at analysis time.

**others** is used to construct the set of associations relying on previous specified associations and a constrained subtype during analysis. It relies on any preceding associations which would make for a high complexity run time should the subtype not be constrained, full of run time checks you wouldn't need with an index constraint.

The result of a 'RANGE attribute can either be locally or globally static. Your usage is not locally static see -2008 9.4.2 Locally static primaries para 1, f) and l), while it is globally static 9.4.3 Globally static primaries para 1, g).). Locally static implies you could produce a sequence of associations during analysis. Globally static implies a loop with parameters defined during elaboration. 

You're viewing this for a special case where **others** is the only choice, which would be fairly simple to implement. The standard takes the view the complexity is unwarranted for those cases where it is not the only choice, perhaps particularly when there is another formal choice method available with the range attribute.

About here you could get the idea that UVVM implementation should have been using the range attribute in the first place.

This is the subject of the -2002 enhancement requiest found in <a href="http://www.eda-twiki.org/isac/IRs-VHDL-2002/IR2065.txt">Issue Request 2065</a>:

>Please find below 3 examples of aggregates that are not allowed in
vhdl vhdl 93-2002. We would like to have it support in the next vhdl
standard. Most of the compilers (like modelsim) already accept those
kind of code if you enable the option to relax the vhdl standard. So I
don't thing that there is any good technical reason to not allow those
kind of codes. CASE1 and CASE2 are really annoying limitations as the
tendencies of today is write more and more generic codes or general
functions/procedures. (note: minor typos fixed)

The idea here being that you should have to explicitly relax the rule found in the sub clause on array aggregates and not doing so insures portability (incidentally adhering to the standard).

Note that  the assertion `Most of the compilers...` is not supported in the enhancement request nor is there any method to indicate change to the standard is imminent lacking a current <a href="http://dlia.ir/Scientific/IEEE/iel1/4499/12755/00587068.pdf">The  Sense  of  VASG</a> (IEEE 1076-CONC-1990) publication. Instead new implementations might be expected to adhere to the standard.

Also note Tristan's comment in the current proposal (Relaxed OTHERS Rules for Aggregates, <a href="http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/RelaxedOthersAggregate#Non_Supporters"> Non-Supporters</a>) in the current proposal.

closed 2017-06-22 by tgingold