ghdl/ghdl #199: Question: body of package "..." was never analyzed

Paebbels, 2016-11-11:
I created a quite complex testbench setup involving PoC's common and simulation packages, an I²C controller and UVVMs I2C verification IP. The UVVM packages compile without errors but with critical warnings such as:
* type of a shared variable must be a protected type
* pure function "..." cannot call (impure) procedure "..."

The complete setup got analyzed. When I start the simulation in GHDL (mcode), I get this error:
> body of package "td_queue_pkg" was never analyzed

**So for now my questions are:**
* What's causing this error (message)?
* How can I narrow it down?

`td_queue_pkg` is an instance of a generic package, so it could be related to the new feature of generic packages. Shall I package the whole testbench as a ZIP or somehow provide it as a Git branch?

tgingold, 2016-11-11:
You can try to narrow it by reducing the size of the design.
Posting (or sending directly to me) a ZIP is OK, but please create a simple script to build it.

tgingold, 2016-11-11:
I think I have a reproducer.

commit 70416572 (2016-11-12): Handle top-level instantiation of macro-expansed packages.

Fix #199
commit 5d3ea061 (2016-12-05): vhdl08: support top-level macro-expanded package instantiation declarations.

Fixes #199
closed 2016-11-12 by tgingold