ghdl/ghdl #200: Does GHDLs documentation provide an intersphing file?

Paebbels, 2016-11-12:
I would like to link some parts of my documentation to GHDLs documentation. Therefore Sphinx offers the intersphinx extension. What's GHDLs URL to the intersphinx inventory file?

tgingold, 2016-11-12:
On 12/11/16 05:46, Patrick Lehmann wrote:
> I would like to link some parts of my documentation to GHDLs
> documentation. Therefore Sphinx offers the intersphinx extension. What's
> GHDLs URL to the intersphinx inventory file?

I am not aware about intersphinx.   Any patch is welcome!

Paebbels, 2016-11-12:
I found that Sphinx creates the `object.inv` in `_build/html`. So it should be accessible at `http://ghdl.readthedocs.io/en/latest/objects.inv`. ([Link](http://ghdl.readthedocs.io/en/latest/objects.inv)) My browser can load a file ...

So this should answer my URL and linking question.

```
# Sphinx inventory version 2
# Project: GHDL
# Version: 0.33
# The remainder of this file is compressed using zlib.
xÚ¥XKoÛ8¾ëWØ^¹@¯{KÛtÀŒ¤h÷FÐ"
```

Hmmm the GHDL documentation is still `0.33`. I'm working on a Python snippet for Sphinx's `conf.py` to extract the current version number from Git (latest tag name). So the Documentation displays the correct version number matching the underlying source code / docs state.

The next question is now, which identifiers does GHDLs documentation provide?
How can I decompress zlib content (as I understood it's stream compressed, so no zip content).

**Two other notes regarding Sphinx and ReadTheDocs:**
* I see you linked the index page `genindex.html` at the bottom of the start page. I found a way to reference it from the navigation bar:

  Workaround steps:
  1. Create a `genindex.rst` in you documentation root folder.
  2. Add a headline like "Index" to the created file.
  3. Reference `genindex` from your `toctree` directive.

  This creates and references a `genindex.html` file in the root directory, which is later overwritten by Sphinx when generating the real index page.

* You can create "parts" (LaTeX terminology / chapter headlines) in your navigation bar, by using multiple `toctree` directives and setting the option `:caption: Part Title`.
  ```rst
  .. toctree::
     :caption: Introduction

     chapter1
     chapter2

  .. toctree::
     :caption: Appendix

     chapter3
     chapter4
  ```

  See the latest [PoC-Library Documentation](http://poc-library.readthedocs.io/en/release/).

Paebbels, 2016-11-13:
Hmmm, I still can't resolve symbols from GHDL. Sphinx prints out that GHDL's inventory file was downloaded, but references gave warnings like this: `index.rst:74: WARNING: unknown option: ghdl:--ieee`

Normally, one could cross-reference with this syntax:
```rst
:option:`ghdl:--ieee`
:option:`GHDL options <ghdl:--ieee>`
```

In contrast to these problems, I can reference classes in the Python documentation (also using Sphinx).
See this example: http://poc-library.readthedocs.io/en/release/PyInfrastructure/PoC.html#PoC.main for a reference to `argparse`.

1138-4EB, 2017-03-01:
Little update:
- This syntax works, and allows to reference and link external sections:
```
:ref:`GHDL Roadmap <ghdl:CHANGE:Roadmap>`
```
- These allow to reference external program options, but no link is created:
```
:ghdl:option:`--ieee`
or
:option:`ghdl:--ieee`
```

1138-4EB, 2017-03-02:
Well, in the end, it was a bug in Sphinx, not anything we were doing wrong. It will be fixed in 1.5.4, see https://github.com/sphinx-doc/sphinx/issues/3487#event-983883880

Meanwhile, we can use the syntax shown in my previous comment, so that no change is required when the new version is released.

I think this issue can be closed now. @Paebbels?

Paebbels, 2017-03-02:
Sometimes new people need to report old bugs ... I already asked for updates on 2822 ...

closed 2017-03-02 by Paebbels