ghdl/ghdl #201: GHDL exception while parsing a generic package with constrained type parameters

Paebbels, 2016-11-14:
I have tried to locally instantiate a OSVVM ScoreBoard package in an architecture:

```VHDL
library ieee;
use     ieee.std_logic_1164.all;
use     ieee.numeric_std.all;

library OSVVM;

entity e is
end entity;

architecture a of e is
  subtype T_DATA		is std_logic_vector(31 downto 0);
  type T_DATA_VECTOR		is array(natural range <>) of T_DATA;

  package ScoreBoardPkg is new osvvm.ScoreboardGenericPkg
    generic map (
      ExpectedType        => T_DATA_VECTOR, --(15 downto 0),
      ActualType          => T_DATA_VECTOR(15 downto 0),  -- this causes the exception
      Match               => match,
      expected_to_string  => to_string,
      actual_to_string    => to_string
    );
begin
end architecture;
```

I'm nor sure if it's allowed to specify a constrained array in this line, but GHDL shouldn't crash :). GHDL analyze results in:

```
sem_name: cannot handle IIR_KIND_ASSOCIATION_ELEMENT_BY_EXPRESSION (D:\git\PoC\tb\sort\sortnet\sortnet_BitonicSort_tb.vhdl:96:61)

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (commit: 2016-11-12;  git branch: paebbels/master';  hash: 7db0826) [Dunoon edition]
Compiled with GNAT Version: GPL 2016 (20160515-49)
In directory: D:\git\PoC\temp\ghdl\
Command line:
C:\Tools\GHDL\0.34dev\bin\ghdl.exe -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=test -PD:\git\PoC\t
emp\precompiled\ghdl D:\git\PoC\tb\sort\sortnet\sortnet_BitonicSort_tb.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : errorout.adb:158
******************************************************************
```

tgingold, 2016-11-14:
On 14/11/16 02:38, Patrick Lehmann wrote:
> I have tried to locally instantiate a OSVVM ScoreBoard package in an
> architecture:
>
> library ieee;
> use     ieee.std_logic_1164.all;
> use     ieee.numeric_std.all;
>
> library OSVVM;
>
> entity e is
> end entity;
>
> architecture a of e is
>   subtype T_DATA		is std_logic_vector(31 downto 0);
>   type T_DATA_VECTOR		is array(natural range <>) of T_DATA;
>
>   package ScoreBoardPkg is new osvvm.ScoreboardGenericPkg
>     generic map (
>       ExpectedType        => T_DATA_VECTOR, --(15 downto 0),
>       ActualType          => T_DATA_VECTOR(15 downto 0),  -- this causes the exception
>       Match               => match,
>       expected_to_string  => to_string,
>       actual_to_string    => to_string
>     );
> begin
> end architecture;

A workaround is to declare the subtype before instantiating the generic:

   subtype t_data_vector_15 is t_data_vector (15 downto 0);

   ..

   actualtype => t_data_vector_15,

Tristan.

tgingold, 2016-12-08:
Now fixed.

closed 2016-12-08 by tgingold