ghdl/ghdl #203: Function parameter in package instantiation: Signature not allowed in expression.

Paebbels, 2016-11-14:
While testing OSVVMs scoreboard, I got this message:
> D:\git\PoC\tb\sort\sortnet\sortnet_BitonicSort_tb.vhdl: 100:56: signature not allowed in expression

This issue might be related to #202.

```VHDL
library ieee;
use     ieee.std_logic_1164.all;

library OSVVM;

entity e is
end entity;

architecture a of e is
  package P_Scoreboard is new OSVVM.ScoreboardGenericPkg
    function foo(val : integer) return string is
    begin
      return integer'image(val);
    end function;

    generic map (
      ExpectedType        => integer,
      ActualType          => integer,
      Match               => "=",
      expected_to_string  => foo[integer return string],  -- this causes the error
      actual_to_string    => foo
    );
  alias T_SCOREBOARD is P_Scoreboard.ScoreBoardPType

  shared variable ScoreBoard : T_SCOREBOARD;
begin

end architecture;
```

----------------------
Distribution list: @JimLewis, @tmeissner

tgingold, 2016-11-14:
On 14/11/16 05:30, Patrick Lehmann wrote:
> While testing OSVVMs scoreboard, I got this message:
>
>     D:\git\PoC\tb\sort\sortnet\sortnet_BitonicSort_tb.vhdl: 100:56:
>     signature not allowed in expression

But that's correct: signatures are not allowed here.
(Ok that should be name instead of expression in the error message).

Tristan.

Paebbels, 2016-11-14:
How or when does VHDL find the correct overload?

JimLewis, 2016-11-14:
Hi Patrick,
The signature is in the formal parameter.  For generics then if you want to use
named association, the names must be different.   Signatures were an oversight
and I think we have it as an ISAC change request.

Jim

>
> How or when does VHDL find the correct overload?

-- 
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Jim Lewis
VHDL Training Expert, SynthWorks
IEEE 1076 VHDL Working Group Chair
Open Source VHDL Verification Methodology (OSVVM), Chief Architect and Co-founder

1-503-590-4787
Jim@SynthWorks.com
http://www.SynthWorks.com

VHDL Training on leading-edge, best coding practices for hardware design and verification.
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

closed 2016-11-23 by tgingold