ghdl/ghdl #204: Warning: pure function "inside" cannot call (impure) interface function "="

Paebbels, 2016-11-14:
I converted the OSVVM package SortListPkg_int to a more generic version. Here is a part of the package, which should reproduce the warning:

```VHDL
package SortListGenericPkg is
  generic (
    type ElementType;
    type ArrayofElementType;
    function array_length(A : ArrayofElementType) return natural;
    function element_get(A : ArrayofElementType; index : natural) return ElementType
  );

  function inside (constant E : ElementType; constant A : in ArrayofElementType) return boolean ;
end package;

package body SortListGenericPkg is
  function inside (constant E : ElementType; constant A : in ArrayofElementType) return boolean is
  begin
    for i in 0 to array_length(A) - 1 loop  --A'range loop
      if E = element_get(A, i) then
        return TRUE ;
      end if ;
    end loop ;
    return FALSE ;
  end function inside ;
end package body;
```

GHDL reports:
> warning: pure function "inside" cannot call (impure) interface function "="
> warning: (interface function "=" is defined here)

The second warning line points to line 3 where the type was given as a generic parameter:
`type ElementType;`

**I have two questions:**
* Why is `"="` called an *interface function*?
  Is it because it's made visible through the generic list?
* Why is this operator function *impure*?

---------------------
Distribution List: @JimLewis, @tmeissner

JimLewis, 2016-11-24:
Not sure, but it is passed implicitly as part of a type - and in this case generic type.  Generally most things done in a protected type are impure.  Certainly anything that calls a function defined in a protected type is impure.  Hence, there is no downside to a function inside of a protected type being impure and the easy path to addressing the warning/error is to change the function inside to impure.   GHDL has helped me find many of this sort of issues.

tgingold, 2016-12-08:
The "=" is called interface function because when you are using interface type, there are implicit interface function for both "=" and "/=".  The "=" referenced is the implicit one.

The operator should be pure.  I have a patch for that.

Thank you for the test case.

commit 9fb8b565 (2016-12-09): Set pure on implicit interface function.

Fix #204
closed 2016-12-09 by tgingold