ghdl/ghdl #205: A operator symbol as a generic function parameter is not yet allowed in a package.

Paebbels, 2016-11-14:
Today I learned from @lievenlemiengre, that it is possible to use operator symbols as a formal in generic lists for function names.

**Example code:**
```VHDL
package SortListGenericPkg is
  generic (
    type ElementType;
    function "<"(L : ElementType; R : ElementType) return boolean;
    function "<="(L : ElementType; R : ElementType) return boolean
  );
end package;
```

GHDL gives this error message:
> incorrect formal name

tgingold, 2016-12-08:
This is in fact accepted.  I suppose your test case is incomplete!

Paebbels, 2016-12-08:
As I tested the code, GHDL didn't accept e.g. `"<"` as function name. I think there was a commit allowing operator symbols at more places...

tgingold, 2016-12-08:
Ok, I am trying to complete the test case.

Paebbels, 2016-12-08:
The simple package declaration is enough plus `ghdl.exe -a --std=08 operator_symbol_problem.vhdl`.

I can confirm that `GHDL 0.34dev (commit: 2016-09-14;  git branch: paebbels/master';  hash: 62f828e)` had this bug (I have no GHDL version in between).

**Error messages:**

    .\operator_symbol_problem.vhdl:4:5: 'signal', 'constant', 'variable', 'file' or identifier expected
    .\operator_symbol_problem.vhdl:4:14: an identifier is expected instead of '<string>'

tgingold, 2016-12-08:
On 08/12/16 04:21, Patrick Lehmann wrote:
> The simple package declaration is enough plus |ghdl.exe -a --std=08
> operator_symbol_problem.vhdl|.
>
> I can confirm that |GHDL 0.34dev (commit: 2016-09-14; git branch:
> paebbels/master'; hash: 62f828e)| had this bug (I have no GHDL version
> in between).

This version is too old I fear.

Paebbels, 2016-12-08:
I can't reproduce it with the current release. So it was fixed. But the code example would be enough to show the old behavior if GHDL wouldn't have fixed it :).

Close?

tgingold, 2016-12-08:
On 08/12/16 04:37, Patrick Lehmann wrote:
> I can't reproduce it with the current release. So it was fixed. But the
> code example would be enough to show the old behavior if GHDL wouldn't
> have fixed it :).
>
> Close?

Not yet, there are some issues at instantiation...

commit 7f6e0383 (2016-12-08): issue #205: add instantiation.

Fix #205
closed 2016-12-08 by tgingold