ghdl/ghdl #207: ghdl 0.33 for Linux x86_64 do not support overload of procedures

mriscoc, 2016-11-18:
ghdl 0.33 for Linux x86_64 executed for syntax check a VHDL package with multiple overloaded procedures:

**ghdl -s pack.vhd**

error messages for each overloaded procedure:

**pack.vhd: .... redeclaration of procedure .... defined at pack.vhd: ...**

On Windows 10 x64 the same package do not show any error.

Paebbels, 2016-11-18:
Hello Miguel,

the 0.33 version is quite outdated. Please use one of the [0.34dev releases](https://github.com/tgingold/ghdl/releases?ts=2) or try to build it locally e.g. with LLVM backend.

tgingold, 2016-11-18:
On 18/11/16 08:05, Miguel Risco Castillo wrote:
> ghdl 0.33 for Linux x86_64 executed for syntax check a VHDL package with
> multiple overloaded procedures:
>
> *ghdl -s pack.vhd*
>
> error messages for each overloaded procedure:
>
> *pack.vhd: .... [...]

That looks very weird.  But without a reproducer, it is impossible to 
investigate.

Note that there is no binary for Windows x64, only for x86.

Tristan.

Paebbels, 2016-11-18:
Oh sorry. I overlooked the Windows 10 hint. I could send you an up-to-date build.

mriscoc, 2016-11-19:
OT updated, thanks.

diogratia, 2016-11-20:
The reproducer fails on ghdl-llvm Commit: afe8072bf5faa1dca1705a08e88818d454ff8758 [afe8072] (28 Sept 2016, ghdl-0.34dev).

```
ghdl -a pack.vhd  
pack.vhd:17:13: redeclaration of procedure "inc" defined at line 14:13  
pack.vhd:18:13: redeclaration of procedure "inc" defined at line 16:13  
pack.vhd:23:13: redeclaration of procedure "dec" defined at line 20:13  
pack.vhd:24:13: redeclaration of procedure "dec" defined at line 22:13  
ghdl: compilation error
```

(This is unchanged since -1993).

SIEEE Std 1076-2008   

**12.1 Declarative region** para 1 c):

>With two exceptions, a declarative region is a portion of the text of the description. A single declarative region is formed by the text of each of the following:  
..  
>>c) A subprogram declaration, together with the corresponding subprogram body

Each subprogram declaration (And it's body) is a declarative region.

**12.5 The context of overload resolution** para 4 b)

>When considering possible interpretations of a complete context, the only rules considered are the syntax rules, the scope and visibility rules, and the rules of the form as follows:  
>...  
>b) Any rule that requires the type of a name or expression to be a type of a certain class; similarly, any rule that requires a certain type to be a discrete, integer, floating-point, physical, universal, or character type.

So in overloading class (variable/signal) can be considered. However:

**12.3 Visibility** para 7:

>A declaration is said to be *hidden* within (part of) an inner declarative region if the inner region contains a homograph of this declaration; the outer declaration is then hidden within the immediate scope of the inner homograph. Each of two declarations is said to be a *homograph* of the other if and only if both declarations have the same designator, and they denote different named entities, and either overloading is allowed for at most one of the two, or overloading is allowed for both declarations and they have the same parameter and result type profile (see 4.5.1).

These have the same parameter and result type, but not class. Overloading is allowed for both (in a pair). Are they homographs?

**4.5.1 General**
Two formal parameter lists are said to have the same *parameter type profile* if and only if they have the same number of parameters, and if at each parameter position the corresponding parameters have the same base type. Two subprograms are said to have the same *parameter and result type profile* if and only if both have the same parameter type profile, and if either both are functions with the same result base type or neither of the two is a function.

So the class of the formal  parameters is not considered here.  They are homographs.

**12.3 Visibility** para 10:

>Two declarations that occur immediately within the same declarative region, other than the declarative region of a block implied by a component instantiation or the declarative region of a generic-mapped package or subprogram equivalent to a package instance or a subprogram instance, ***shall not be homographs***, unless exactly one of them is the implicit declaration of a predefined operation or is an implicit alias of such an implicit declaration. In such cases, a predefined operation or alias thereof is always hidden by the other homograph. Where hidden in this manner, an implicit declaration is hidden within the entire scope of the other declaration (regardless of which declaration occurs first); the implicit declaration is visible neither by selection nor directly. For a declarative region of a block implied by a component instantiation or the declarative region of a generic-mapped package or subprogram equivalent to a package instance or a subprogram instance, the rules of this paragraph are applied to the corresponding entity declaration, component declaration, uninstantiated package declaration, or uninstantiated subprogram declaration, as appropriate.

And that tells us why ghdl should find it illegal.  Essentially class isn't considered in determining whether two declarations are homographs and the two declarations aren't allowed in the same declarative region.

The solution would be to have two packages where each package declaration is a separate declarative regions (**12.1 Declarative region** para 1 d)), homographs are separated by package and class is considered in the context of overload resolution.

However: 

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package pack0 is

  procedure inc(signal val:inout std_logic_vector);
  procedure inc(signal val:inout unsigned);
  procedure inc(signal val:inout signed);
  procedure inc(signal val:inout integer);
  procedure dec(signal val:inout std_logic_vector);
  procedure dec(signal val:inout unsigned);
  procedure dec(signal val:inout signed);
  procedure dec(signal val:inout integer);
  
end pack0;

[... 136 lines cut ...]
pack1.vhdl:122:12: possible interpretations are:
pack1.vhdl:70:13: procedure dec [unsigned]
pack1.vhdl:12:13: procedure dec [unsigned]
pack1.vhdl:123:12: can't resolve overload for procedure call
pack1.vhdl:123:12: possible interpretations are:
pack1.vhdl:71:13: procedure dec [integer]
pack1.vhdl:14:13: procedure dec [integer]
pack1.vhdl:124:12: can't resolve overload for procedure call
pack1.vhdl:124:12: possible interpretations are:
pack1.vhdl:71:13: procedure dec [integer]
pack1.vhdl:14:13: procedure dec [integer]
ghdl: compilation error
```

[... cut ...] eclaration, to determine that the occurrence is not legal at the given point.  
— The visibility rules determine *more than one* possible meaning. In such a case, **the occurrence of the identifier is legal at this point if and only if *exactly one* visible declaration is acceptable for the overloading rules in the given context or all visible declarations denote the same named entity**.

The work around for that is to use selected names.

So you've uncovered a bug. Just not the bug you thought.

tgingold, 2016-11-20:
On 20/11/16 21:33, Dio Gratia wrote:
> The reproducer fails on ghdl-llvm Commit: afe8072
> <https://github.com/tgingold/ghdl/commit/afe8072bf5faa1dca1705a08e88818d454ff8758>
> afe8072 <28%20Sept%202016,%20ghdl-0.34dev>.

[...]

> ghdl is not using class in determining overload resolution here (despite
> *12.5 The context of overload resolution* para 4 b)).

Isn't 12.5 4 b) about class of type as defined in 5.1 rather than object 
class (as defined in 6.4.1) ?

Tristan.

diogratia, 2016-11-21:
I had to go off and police the paddocks with my trusty pitchfork and wheelbarrow.  While out and about I realized I had made the same mistake some years ago. ghdl appears to be doing the right thing no matter where the subprogram declarations are found.

I also tried part.vhd and part1.vhd in @nickg 's nvc. part.vhd gets the errors. part1.vhdl gets an elaboration failure (diagnostics sans error message) for entity overload.  (And you'd think visibility issues would be detected at analysis time).

closed 2016-11-23 by tgingold

mriscoc, 2016-11-27:
I want to take the opportunity to say thank you for your great work with GHDL, I'm using it with a multi platform tool to generate a simple bus architecture for FPGA: http://sba.accesus.com/software-tools/sba-creator

Regarding the commented bug, I use a library with multiple overloading functions and procedures in several tools from Xilinx, Microsemi (Actel) and Altera. The errors about the overloads show up only in GHDL.

Regards

tgingold, 2016-11-28:
From the discussion, GHDL is correct while there is a bug in Xilinx, Actel and Altera tools.  You can report it (they certainly use the same vhdl frontend).

Did you try with a commercial simulator like Modelsim ?