ghdl/ghdl #209: New type defined in package: cant use operator &

petspats, 2016-11-23:
In file util.vhd I defined a new array type:
``` vhdl
package Util is
  type integer_list_t is array (natural range <>) of integer;
end package;
```

Trying to use it in another file:
``` vhdl
library work;
    use work.all;

package ShiftReg is
    procedure main(new_sample: integer);
end package;

package body ShiftReg is
    procedure main(new_sample: integer) is
        variable dummy: Util.integer_list_t(0 to 3); -- Here i use the type
    begin
        dummy := new_sample & dummy(0 to dummy'high-1); -- Error about missing &
    end procedure;
end package body;
```

Error:
```
ghdl --version
GHDL 0.34dev (20151126) [Dunoon edition]
 Compiled with GNAT Version: 4.6

ghdl -a --std=08 --work=work util.vhd main.vhd
/tmp/tmpy2fliixl/main.vhd:12:29: no function declarations for operator "&"
```

If i move type declaration to current package (No more Util package):
```vhdl
library work;
    use work.all;

package ShiftReg is
    type integer_list_t is array (natural range <>) of integer; -- notice this line
    procedure main(new_sample: integer);
end package;

package body ShiftReg is

    procedure main(new_sample: integer) is
        variable dummy: integer_list_t(0 to 3); -- notice this line
    begin
        dummy := new_sample & dummy(0 to dummy'high-1); --no error
    end procedure;

end package body;
```
`ghdl -a --std=08 --work=work main.vhd`
All works.

tgingold, 2016-11-23:
Ghdl looks right here.  You need to write: use work.Util.all;

petspats, 2016-11-23:
@tgingold Works. Fastest support i have experienced, thanks :)

closed 2016-11-23 by petspats

tgingold, 2016-11-23:
On 23/11/16 20:28, petspats wrote:
> @tgingold <https://github.com/tgingold> Works. Fastest support i have
> experienced, thanks :)

You were lucky, I was reading my mails.