ghdl/ghdl #210: BUILD.txt correction

markNZed, 2016-11-29:
Line 89: cd ..
Should be: cd /path/to/ghdl
Or maybe gcc source is expected to be in the ghdl dir, then line 66 is misleading

I've been setting up the Dalton i8051 to run on ghdl. Let me know if the source for that is of interest.

tgingold, 2016-11-29:
I think BUILD.txt is correct as is.  On line 66 you untar gcc sources, but you build gcc in a subdirectory of ghdl (line 77).  Maybe a 'cd /path/to/ghdl' should be added just before line 66 ?

And yes, the sources are interesting.

markNZed, 2016-11-29:
Line 57 
./configure --with-gcc=/path/to/gcc/source/dir --prefix=/usr/local

seems to indicate a path to gcc outside of the ghdl tree. I would expect something like:

./configure --with-gcc=./gcc_source_dir --prefix=/usr/local

Maybe it is just me, but I untar the gcc sources next to the ghdl sources so things got confusing after that. I think it was the absolute path to gcc source dir that lead me to understand I could put the gcc sources anywhere.

If I understand you, the make in line 86 will be in ghdl/gcc/gcc-objs so the "cd .." in line 89 would take us to ghdl/gcc instead of ghdl dir ?

I could package the source and scripts up in a tar and share this on Dropbox. Does that work for you ?

tgingold, 2016-11-29:
On 29/11/16 20:53, markNZed wrote:
> Line 57
> ./configure --with-gcc=/path/to/gcc/source/dir --prefix=/usr/local
>
> seems to indicate a path to gcc outside of the ghdl tree.

This is correct: gcc sources can be outside of the ghdl tree.

 > I would expect something like:
>
> ./configure --with-gcc=./gcc_source_dir --prefix=/usr/local
>
> Maybe it is just me, but I untar the gcc sources next to the ghdl
> sources

That's fine.

 > so things got confusing after that.

Why ?

> I think it was the absolute
> path to gcc source dir that lead me to understand I could put the gcc
> sources anywhere.

Yes, you can put the gcc sources anywhere.

> If I understand you, the make in line 86 will be in ghdl/gcc/gcc-objs
 > so
> the "cd .." in line 89 would take us to ghdl/gcc instead of ghdl dir ?

Ah, ok.  You're right.  Sorry, I was confused.

> I could package the source and scripts up in a tar and share this on
> Dropbox. Does that work for you ?

No, it's ok.  Indeed line 89 should be changed.

Thanks
Tristan.

markNZed, 2016-11-29:
For the Dropbox link I meant the sources and scripts for the i8051 design and testbench.

tgingold, 2016-12-04:
Is it from http://www.cs.ucr.edu/~dalton/i8051/i8051syn/ ?

markNZed, 2016-12-04:
Yes that's it.

commit 01ae3f0d (2016-12-05): Clarify BUILD instructions with gcc.

Close #210.
closed 2016-12-05 by tgingold