ghdl/ghdl #211: VCD replay

markNZed, 2016-11-30:
I don't see an option for ghdl to replay a waveform file to generate stimuli, Modelsim has an option -vcdstim for this. It would be a nice feature to have.

Is there any easy way to drive a design with a waveform file in ghdl ? 

I guess one option would be to generate a VHDL testbench and read the waveform file directly or transform the waveform into a simpler format for the testbench to read.

An unrelated question  - is there any online chat e.g. IRC support for ghdl ?

Paebbels, 2016-11-30:
Hello,

replaying a waveform is out of the scope of a simulator. It's either a waveform viewer task (-> e.g. GTKWave) or a testbench task to consume external input data. The latter can be implemented in a VHDL package.

Speaking of this, I have a VCD Parser written in VHDL, which can drive a testbench. Shall I provide it?
It might need some redesign improvements, because it's quite old and was a fast development to read data from ChipScope (measured real data and used for a fine grained simulation to investigate a bug).

GHDL has a Gitter chat room: https://gitter.im/ghdl1/Lobby.
You can use your GitHub account to login.

Kind regards
    Patrick

markNZed, 2016-11-30:
Hi Patrick, I think it makes sense to keep it out of scope even if some simulators do provide this. I'd be very interested in your VHDL package for VCD parsing  - it is a great idea. I was thinking of using https://metacpan.org/pod/Verilog::VCD and generating the test harness. I'll switch over to Gitter now.

closed 2016-12-05 by tgingold

Paebbels, 2016-12-05:
Just for completeness...

The mentioned VCD parser is now released in [The PoC-Library](https://github.com/VLSI-EDA/PoC?ts=2) in the namespace `sim` as [`sim_VCDParser`](https://github.com/VLSI-EDA/PoC/blob/release/src/sim/sim_VCDParser.vhdl?ts=2) package. A [simple testbench](https://github.com/VLSI-EDA/PoC/blob/release/tb/sim/sim_VCDParser_tb.vhdl?ts=2) is also available and can be tested with e.g. GHDL.

```
PS> .\poc.ps1 ghdl PoC.sim.VCDParser -g
```

The resulting output should look like this:
[image]