ghdl/ghdl #212: are_trees_equal internal error

ljjterma, 2016-12-02:
The following code:

`
PACKAGE test_pkg IS

  TYPE test_record_t IS RECORD
    number : integer;
  END RECORD test_record_t;

  FUNCTION set_test_record_default
    RETURN test_record_t;

  FUNCTION set_test_record (
    CONSTANT C_TEST : test_record_t := set_test_record_default)
    RETURN test_record_t;

END PACKAGE test_pkg;

PACKAGE BODY test_pkg IS

  FUNCTION set_test_record_default
    RETURN test_record_t IS
    VARIABLE result : test_record_t;
  BEGIN
    result.number := 0;
    RETURN result;
  END set_test_record_default;

  FUNCTION set_test_record (
    CONSTANT C_TEST : test_record_t := set_test_record_default)
    RETURN test_record_t IS
  BEGIN
    RETURN C_TEST;
  END set_test_record;

END PACKAGE BODY test_pkg;

ENTITY test IS
END ENTITY test;

LIBRARY work;
USE work.test_pkg.set_test_record;

ARCHITECTURE rtl OF test IS

BEGIN

END ARCHITECTURE rtl;
`

Generates the internal error:

are_trees_equal: cannot handle IIR_KIND_FUNCTION_CALL (test.vhd:28:40)

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: GPL 2016 (20160515-49)
In directory: /home/ljj/work_SBL-1012/hw/hdl/test/
Command line:
ghdl -m test
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : errorout.adb:168
Call stack traceback locations:
0x4e7895 0x67718b 0x6768af 0x67658a 0x676781 0x6771be 0x678084 0x673467 0x67998f 0x67a7f2 0x583ea3 0x584728 0x56d663 0x56da2a 0x56d6de 0x56d6de 0x56eed4 0x5778e0 0x63888f 0x634dfb 0x4cd9c6 0x4053d7 0x7fd037819b13 0x404257 0xfffffffffffffffe
******************************************************************

Ghdl is build from source on a CentOS 7.2 Linux machine.

Paebbels, 2016-12-02:
Hello @ljjterma,

please use [fended code blocks](https://help.github.com/articles/creating-and-highlighting-code-blocks/#syntax-highlighting) to format your VHDL code example.

tgingold, 2016-12-03:
On 02/12/16 08:23, ljjterma wrote:
> The following code:
[...]

> Generates the internal error:
>
> are_trees_equal: cannot handle IIR_KIND_FUNCTION_CALL (test.vhd:28:40)

I will fix it, but I am not sure you can have function calls in default 
value for parameters.  The grammar says: := /static/_expression
but that is not explained in the text.

Thank you for this report,
Tristan.

Paebbels, 2016-12-03:
I think I have seen such code, **but** the initializing function (in your example `set_test_record_default`) was defined in another package, which was already analyzed.

The function `set_test_record` is going to be analyzed and requires a default value, but the body of `set_test_record_default` is not yet analyzed. This might be a problem.

ljjterma, 2016-12-05:
The set_test_record_default is defined right before set_test_record. In the package, you only need the return type of set_test_record_default. Not the body.

commit 927f45f7 (2016-12-07): are_trees_equal: handle function_call

Fixes #212
closed 2016-12-07 by tgingold