ghdl/ghdl #213: Has GCCs -fPIC behavior changed?

Paebbels, 2016-12-02:
Hello,

I did ne fresh GHDL build and executed a simple testbench. I get the following output. Please see the second half (elaboration), where `/usr/bin/ld` complains about `-fPIC`.

```Bash
paebbels@debian8:~/git/PoC$ ./poc.sh -v ghdl arith.prng
================================================================================
                         The PoC-Library - Service Tool                         
================================================================================
Initializing PoC-Library Service Tool for simulations
  Reading configuration files...
Preparing simulation environment...
  Creating fresh temporary directory.
  Changing working directory to temporary directory.
  Preparing GHDL simulator.
Testbench: PoC.arith.prng
  Creating PoC project 'arith_prng_tb'
  Reading filelist '/home/paebbels/git/PoC/tb/arith/arith_prng_tb.files'
Running analysis for every vhdl file...
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=poc /home/paebbels/git/PoC/tb/common/my_project.vhdl
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=poc /home/paebbels/git/PoC/tb/common/my_config_GENERIC.vhdl
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=poc /home/paebbels/git/PoC/src/common/utils.vhdl
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=poc /home/paebbels/git/PoC/src/common/config.vhdl
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=poc /home/paebbels/git/PoC/src/common/math.vhdl
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=poc /home/paebbels/git/PoC/src/common/strings.vhdl
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=poc /home/paebbels/git/PoC/src/common/vectors.vhdl
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=poc /home/paebbels/git/PoC/src/common/physical.vhdl
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=poc /home/paebbels/git/PoC/src/common/components.vhdl
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=poc /home/paebbels/git/PoC/src/common/protected.v08.vhdl
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=poc /home/paebbels/git/PoC/src/common/fileio.v08.vhdl
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=poc /home/paebbels/git/PoC/src/sim/sim_types.vhdl
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=poc /home/paebbels/git/PoC/src/sim/sim_random.v08.vhdl
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=poc /home/paebbels/git/PoC/src/sim/sim_protected.v08.vhdl
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=poc /home/paebbels/git/PoC/src/sim/sim_global.v08.vhdl
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=poc /home/paebbels/git/PoC/src/sim/sim_simulation.v08.vhdl
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=poc /home/paebbels/git/PoC/src/sim/sim_waveform.vhdl
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=poc /home/paebbels/git/PoC/src/arith/arith.pkg.vhdl
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=poc /home/paebbels/git/PoC/src/arith/arith_prng.vhdl
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=test /home/paebbels/git/PoC/tb/arith/arith_prng_tb.vhdl
Running elaboration...
  command: /opt/ghdl/0.34-dev-llvm/bin/ghdl -e -fexplicit --ieee=synopsys --std=08 --work=test arith_prng_tb
  ghdl elaborate messages for 'test.arith_prng_tb'
  ------------------------------------------------------------------------------
  /usr/bin/ld: e~arith_prng_tb.o: relocation R_X86_64_32 against `.bss' can not be used when making a shared object; recompile with -fPIC
  /usr/bin/ld: /opt/ghdl/0.34-dev-llvm/lib/ghdl/v08/std/std_standard.o: relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC
  /usr/bin/ld: /opt/ghdl/0.34-dev-llvm/lib/ghdl/v08/std/textio.o: relocation R_X86_64_32 against symbol `std__textio__RTI' can not be used when making a shared object; recompile with -fPIC
  /usr/bin/ld: /opt/ghdl/0.34-dev-llvm/lib/ghdl/v08/std/textio_body.o: relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC
  /usr/bin/ld: /opt/ghdl/0.34-dev-llvm/lib/ghdl/v08/synopsys/std_logic_1164.o: relocation R_X86_64_32 against symbol `ieee__std_logic_1164__RTI' can not be used when making a shared object; recompile with -fPIC
  /usr/bin/ld: /opt/ghdl/0.34-dev-llvm/lib/ghdl/v08/synopsys/std_logic_1164-body.o: relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC
  /usr/bin/ld: /opt/ghdl/0.34-dev-llvm/lib/ghdl/v08/synopsys/numeric_std.o: relocation R_X86_64_32 against symbol `ieee__numeric_std__RTI' can not be used when making a shared object; recompile with -fPIC
  /usr/bin/ld: /opt/ghdl/0.34-dev-llvm/lib/ghdl/v08/synopsys/numeric_std-body.o: relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC
  /usr/bin/ld: /opt/ghdl/0.34-dev-llvm/lib/ghdl/v08/synopsys/math_real.o: relocation R_X86_64_32 against symbol `ieee__math_real__RTI' can not be used when making a shared object; recompile with -fPIC
  /usr/bin/ld: /opt/ghdl/0.34-dev-llvm/lib/ghdl/v08/synopsys/math_real-body.o: relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC
  /usr/bin/ld: utils.o: relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC
  /usr/bin/ld: my_config_GENERIC.o: relocation R_X86_64_32 against symbol `poc__my_config__RTI' can not be used when making a shared object; recompile with -fPIC
  /usr/bin/ld: my_project.o: relocation R_X86_64_32 against symbol `poc__my_project__RTI' can not be used when making a shared object; recompile with -fPIC
  /usr/bin/ld: config.o: relocation R_X86_64_32S against symbol `poc__config_private__t_board_config_string__STB' can not be used when making a shared object; recompile with -fPIC
  /usr/bin/ld: strings.o: relocation R_X86_64_32S against symbol `poc__strings__t_rawchar__STB' can not be used when making a shared object; recompile with -fPIC
[... cut ...]
Time:  0:02  Count: 1    Passed: 0    No Asserts: 0   Failed: 0   Errors: 1 
================================================================================
```

I know GCC has two PIC options: `-fPIC` and `-fpic`. Has something chainged in GHDL's makefiles, GCC's default PIC behavior or in GNAT?

GHDL version:
```Bash
paebbels@debian8:~/git/PoC$ /opt/ghdl/0.34-dev-llvm/bin/ghdl -v
GHDL 0.34-dev (2016-09-14-222-g7db0826-dirty) [Dunoon edition]
 Compiled with GNAT Version: 6.2.0
 llvm code generator
```

tgingold, 2016-12-03:
On 02/12/16 20:01, Patrick Lehmann wrote:
> Hello,
>
> I did ne fresh GHDL build and executed a simple testbench. I get the
> following output. Please see the second half (elaboration), where
> |/usr/bin/ld| complains about |-fPIC|.

Did you upgrade your system recently ?

Try to link with -no-pie

Regard,
Tristan.

Paebbels, 2016-12-03:
Hmmm, my test system is a Debian testing, it automatically uses new software releases. Unfortunately, they declared GCC 6.2.0-13 as the new default and jumped from 5.4 over 6.0 (and 6.1?) directly to 6.2 with a zero in the release number ...

**So I'll use this GHDL elaborate call:**
```Bash
/opt/ghdl/0.34-dev-llvm/bin/ghdl -e -v -fexplicit -Wl,-no-pie --ieee=synopsys --std=08 --work=test arith_prng_tb
```

OK this works.

**Some questions remain:**
* Is this a workaround or solution?
* What back ends are effected?
* Should GHDL compile with `-fPIC` by default?

tgingold, 2016-12-04:
On 03/12/16 17:27, Patrick Lehmann wrote:
> Hmmm, my test system is a Debian testing, it automatically uses new
> software releases. Unfortunately, they declared GCC 6.2.0-13 as the new
> default and jumped from 5.4 over 6.0 (and 6.1?) directly to 6.2 with a
> zero in the release number ...
>
> *So I'll use this GHDL elaborate call:*
>
> /opt/ghdl/0.34-dev-llvm/bin/ghdl -e -v -fexplicit -Wl,-no-pie --ieee=synopsys --std=08 --work=test arith_prng_tb
>
> OK this works.
>
> *Some questions remain:*
>
>   * Is this a workaround or solution?

Only a work-around.

>   * What back ends are effected?

I suppose that both llvm and gcc back ends are affected.

>   * Should GHDL compile with |-fPIC| by default?

Yes.  What is the output of 'gcc -v' on your machine ?

Paebbels, 2016-12-04:
The GCC version output is:

```
paebbels@debian8:~/git/PoC$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/6/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 6.2.0-13' --with-bugurl=file:///usr/share/doc/gcc-6/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-6 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-6-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-6-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-6-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 6.2.0 20161109 (Debian 6.2.0-13) 
```

tgingold, 2016-12-05:
On 04/12/16 21:34, Patrick Lehmann wrote:
> The GCC version output is:

So, there is --enable-default-pie
And you are using the llvm backend for ghdl.

Paebbels, 2016-12-05:
Yes the "productive" GHDL version is using LLVM. Sometime I also run tests with the mcode backend. I don't use GCC any more. It's to much work to compile it and GCC itself :)

tgingold, 2016-12-05:
On 05/12/16 02:13, Patrick Lehmann wrote:
> Yes the "productive" GHDL version is using LLVM. Sometime I also run
> tests with the mcode backend. I don't use GCC any more. It's to much
> work to compile it and GCC itself :)

So I think you have to change one line in
  src/ortho/llvm/ortho_code_main.adb:

  Reloc : constant RelocMode := RelocDefault;

-->

  Reloc : constant RelocMode := RelocPic;

Tell me if that works.

Paebbels, 2016-12-05:
I'm looking forward to test it.

Paebbels, 2016-12-05:
The `-no-pie` flag breaks our Travis runs ...

The GCC on Travis-CI is of version 4.8, which doesn't know anything about PIE. My problem is that I could enabled workarounds based on the GHDL version (which is always 0.34-dev), but not on the GCC version with which GHDL was build.

Just in case you need the current `gcc -v` results:
```
Testing GCC version and configuration...
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.8/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 4.8.4-2ubuntu1~14.04.3' --with-bugurl=file:///usr/share/doc/gcc-4.8/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.8 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.8 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libmudflap --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.8-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.3) 

Testing GHDL version...
GHDL 0.34dev (20151126) [Dunoon edition]
 Compiled with GNAT Version: 4.8
 llvm code generator
Written by Tristan Gingold.
Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
GHDL test [SUCCESSFUL]
```
Travis-CI report: https://travis-ci.org/VLSI-EDA/PoC/jobs/181299337#L339

mzabeltud, 2016-12-05:
I have tested `RelocPIC` with GCC 4.9 on Debian which can also generate Position Independent Exceutables (PIE), but it is no the default there.

At first, I changed in file src/ortho/llvm/ortho_code_main.adb the line 73:

```
 Reloc : constant RelocMode := RelocDefault;
```

to

```
 Reloc : constant RelocMode := RelocPIC;
```

as suggested by @tgingold.

Now, the object files generated by LLVM have the neccessary relocation information.

But, the object files of libgrt.a must be position independ too. Thus, one has to change in the main Makefile.in, the line 72:

```
GRT_FLAGS=$(OPT_FLAGS)
```

to

```
GRT_FLAGS=$(OPT_FLAGS) -fPIC
```

Now, the LLVM GHDL simulator binary can be linked during elaboration with option `-Wl,-pie` and also without this option.

----

@Paebbels tested this with GCC 6.2 on Debian. There the linker option '-pie' is set by default.

commit 0710ab89 (2016-12-07): Use -fpic if gcc is configured with --enable-default-pie

Test in configure and set Default_Pie in default_pathes.ads
Pass -fpic to ghdl1 if Default_Pie is set.
Decode -fpic in ghdl1_llvm

Fix #213
closed 2016-12-07 by tgingold