ghdl/ghdl #216: Error: choise is not locally static

ljjterma, 2016-12-08:
The following code:

```

ENTITY test IS
END ENTITY test;

ARCHITECTURE rtl OF test IS

  TYPE test_type_t IS (
    TEST_1,
    TEST_2);

  TYPE test_type_value_t IS ARRAY (test_type_t) OF natural;

  CONSTANT C_TEST_TYPE_VALUE : test_type_value_t := (
    TEST_1 => 0,
    TEST_2 => 1);

  CONSTANT C_TEST_TYPE_TEST_1 : natural := C_TEST_TYPE_VALUE(TEST_1);
  CONSTANT C_TEST_TYPE_TEST_2 : natural := C_TEST_TYPE_VALUE(TEST_2);

  FUNCTION get_priority (
    arg : natural)
    RETURN natural IS
    VARIABLE result_v : natural;
  BEGIN
    CASE arg IS
      WHEN C_TEST_TYPE_TEST_1 =>
        result_v := 3;
      WHEN C_TEST_TYPE_TEST_2 =>
        result_v := 0;
      WHEN OTHERS =>
        REPORT "Unknown Sector Type"
          SEVERITY error;
        result_v := 4;
    END CASE;
    RETURN result_v;
  END FUNCTION get_priority;

BEGIN

END ARCHITECTURE rtl;

```

Generates the error: choise is not locally static

It works fine in Questasim, Vivado and Precision

tgingold, 2016-12-08:
Ghdl is correct: before vhdl-2008, indexed names were not locally static expressions.
This has changed in vhdl-2008, but that's not yet implemented in ghdl.

ljjterma, 2016-12-09:
Will you fix it soon? I am in the process of evaluating ghdl on a large design and errors are show stoppers. Currently I am using Questasim, but I would really like to use ghdl. If I were to fix the issue myself, what would be the easiest way to get started? Do you have any documentation on your preferred tools and the structure of the source code?  Do you have any unit tests of the source code?

Paebbels, 2016-12-09:
Hello,

I'm not the developer but I can answer at least one of your questions:
Testcases can be found in `testsuite` and are executed on every pull request via Travis-CI. They are made of Bash scripts and thus can be run on your local machine without any extra tools.

tgingold, 2016-12-09:
On 09/12/16 08:06, ljjterma wrote:
> Will you fix it soon? I am in the process of evaluating ghdl on a large
> design and errors are show stoppers. Currently I am using Questasim, but
> I would really like to use ghdl.

Interesting to know.  Do you have a list of issues ?  If this is the 
only one issue, it would be worth doing the effort.  If there is a long 
list of issues, that would be different.

> If I were to fix the issue myself, what
> would be the easiest way to get started? Do you have any documentation
> on your preferred tools and the structure of the source code?

This is not a bug but a feature to add.  Are you specifically using vhdl 
2008 ?

You change to change the
expr_staticness of aggregates (see sem_expr.adb) and then to change
evaluation.adb so that sliced names and indexed names can be
evaluated.  This is not a huge amount of work, but more than a couple of 
lines.

 > Do you have any unit tests of the source code?

As written by Patrick, there are regression tests and tests against the 
standard.

Regards,
Tristan.

ljjterma, 2016-12-10:
I am using 2008 when using ghdl. My production code use 93 dialect, but my test benches are mainly 2008 . 

I do not have a huge list of issues. As of now, i have stoppet when i found a problem. 

My experience is that it is a huge benifit to Unit test the code (http://libre.adacore.com/tools/aunit/) . If i were to Fix the issue, i will add Unit tests. I plan to put them in ghdl/src/test. 

Best regards
Lars

tgingold, 2016-12-10:
On 10/12/16 09:40, ljjterma wrote:
> I am using 2008 when using ghdl. My production code use 93 dialect, but
> my test benches are mainly 2008 .
>
> I do not have a huge list of issues. As of now, i have stoppet when i
> found a problem.

OK.  I will try to work on your issue, but I have a couple of issues to 
fix before it.

> My experience is that it is a huge benifit to Unit test the code
> (http://libre.adacore.com/tools/aunit/) . If i were to Fix the issue, i
> will add Unit tests. I plan to put them in ghdl/src/test.

End-to-end tests have a main advantage: they don't need to be maintained 
when the program change and are faster to run.
As ghdl is a compiler, it is not very difficult to test the internals 
through vhdl sources.  The most difficult part to test is the backend, 
so there is the possibility to read intermediate code just to test the 
backend.

YMMV!

Regards,
Tristan.

commit cadb1e20 (2016-12-20): Evaluation: handle array aggregate and indexed names.

Fix #216
closed 2016-12-20 by tgingold