ghdl/ghdl #220: The fPIC problem is still present when build with LLVM-3.8

Paebbels, 2016-12-11:
I saw that it's possible to build GHDL with LLVM-3.8, so I did two builds in disjunctive directories:
- /build/llvm-3.5  
  ```
  mkdir llvm-3.5
  cd llvm-3.5
  ../../configure --prefix=/opt/ghdl/0.34-dev-llvm --with-llvm-config=llvm-config-3.5
  make
  sudo make install
  ```
- /build/llvm-3.8 =>
  ```
  mkdir llvm-3.8
  cd llvm-3.8
  ../../configure --prefix=/opt/ghdl/0.34-dev-llvm --with-llvm-config=llvm-config-3.8
  make
  sudo make install
  ```

When I run our PoC test, I get no errors for the LLVM-3.5 build, but linker errors for the LLVM-3.8 build. The messages are again related to `-fPIC`. Have we missed something that is different in LLVM-3.8?

Compiler is GCC/GNAT 6.2.1.

tgingold, 2016-12-11:
Not expected.  What is the content of default_pathes.ads in the failure case ?

Paebbels, 2016-12-11:
Both `default_pathes.ads` files are identical.

```ADA
package Default_Pathes is

   --  Accept long lines.
   pragma Style_Checks ("M999");

   Install_Prefix : constant String :=
     "/opt/ghdl/0.34-dev-llvm";
   Lib_Prefix : constant String :=
     "lib/ghdl";

   Compiler_Gcc : constant String :=
     "ghdl1-gcc";
   Compiler_Mcode : constant String :=
     "ghdl1-mcode";
   Compiler_Llvm : constant String :=
     "bin/ghdl1-llvm";
   Compiler_Debug : constant String :=
     "ghdl1-debug";
   Post_Processor : constant String :=
     "oread-llvm";

   Shared_Library_Extension : constant String :=
     ".so";
end Default_Pathes;
```

tgingold, 2016-12-11:
On 11/12/16 07:36, Patrick Lehmann wrote:
> Both |default_pathes.ads| files are identical.

Ah ah.  I think you must get the latest pull,
and reconfigure (the configure script has changed).

Tristan.

Paebbels, 2016-12-11:
I'm at 11b88a9. Except for the three latest merges. Everything is in my branch and I rerun configure as illustrated in the initial post. That should overwrite the file, right?

tgingold, 2016-12-11:
On 11/12/16 07:43, Patrick Lehmann wrote:
> I'm at 11b88a9
> <https://github.com/tgingold/ghdl/commit/11b88a9ce1e48c3019c7dd4910a73e76ee33f08a>.
> Except for the three latest merges.

That's ok.

 > Everything is in my branch and I
> rerun configure as illustrated in the initial post. That should
> overwrite the file, right?

Yes, that should overwrite it.

It is created from ghdldrv/default_pathes.ads.in whose last lines are:

    Shared_Library_Extension : constant String :=
      "@SOEXT@";
    Default_Pie : constant Boolean := "@default_pie@" = String'("yes");
end Default_Pathes;

I can't see why Default_Pie won't be present.

Paebbels, 2016-12-11:
I think it's a Git issue ... There are some circumstances when Git dies or aborts a command an leaves the directory in a inconsistent state. I'll reset every single file...

tgingold, 2016-12-11:
On 11/12/16 07:52, Patrick Lehmann wrote:
> I think it's a Git issue ... There are some circumstances when Git dies
> or aborts a command an leaves the directory in a inconsistent state.
> I'll reset every single file...

git checkout -f ?

Paebbels, 2016-12-11:
Now I have this additional line in `default_pathes.ads`:

```ADA
Default_Pie : constant Boolean := "no" = String'("yes");
```

tgingold, 2016-12-11:
On 11/12/16 08:14, Patrick Lehmann wrote:
> Now I have this additional line in |default_pathes.ads|:
>
> Default_Pie : constant Boolean := "no" = String'("yes");

Ok, that's good.  But this means that -fPIC is not required on your 
system.  What is the output of 'gcc -v' ?

Paebbels, 2016-12-11:
`gcc -v` shows

```
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/6/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 6.2.1-5' --with-bugurl=file:///usr/share/doc/gcc-6/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-6 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-6-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-6-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-6-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc=auto --enable-multiarch --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 6.2.1 20161124 (Debian 6.2.1-5) 
```

Paebbels, 2016-12-11:
Can you add a new matrix setting in Travis to also build with LLVM-3.8 and also with newer GCC versions?

tgingold, 2016-12-11:
On 11/12/16 08:20, Patrick Lehmann wrote:
> |gcc -v| shows

Ah, I see.  In configure, that should be 'gcc -v' instead of 'gcc --version'

My fault.

tgingold, 2016-12-11:
On 11/12/16 08:28, Patrick Lehmann wrote:
> Can you add a new matrix setting in Travis to also build with LLVM-3.8
> and also with newer GCC versions?

Make sense.  Now I can even assign it to you :-)

Paebbels, 2016-12-11:
I did a local check. Yes it solved it.

tgingold, 2016-12-11:
On 11/12/16 08:28, Patrick Lehmann wrote:
> Can you add a new matrix setting in Travis to also build with LLVM-3.8
> and also with newer GCC versions?

On ubuntu precise, we can use:
gnat-6 or gnat-5 (or even gnat-5:i386)
llvm-3.8

I will try soon (hopefully today)

tgingold, 2016-12-11:
gcc-6 and llvm-3.8 are not available on 'precise' container.

tgingold, 2016-12-11:
Should be fixed.

closed 2016-12-11 by tgingold

tgingold, 2016-12-12:
Finally, there is a gnat-6 + lvm-3.8 build in travis-ci

Paebbels, 2016-12-12:
That's good to hear.