ghdl/ghdl #222: Logging sim output to a file

markNZed, 2016-12-11:
Hi,

Is there an option to save simulator output to a file ?

Does any simulation output got to STDERR ?

Thanks.

Paebbels, 2016-12-11:
Hello Mark,

how about `ghdl .... &2>1 > lofile.log`?
This captures all (STDOUT, STDERR) output streams and redirects them to a file.

I don't think GHDL needs to provide such a feature (but can be considered a "enhancement") because it's already a capability of your shell (Bash, Cmd, PowerShell, ...).

markNZed, 2016-12-12:
I agree it is possible using various shell features. I'm not sure the above works in all environments. So it might be a nice feature for portable scripts. A minor enhancement.

It also seems that assertions are going to stderr. They may not be buffered either while stdout seems to be buffered. I'm not sure about assertions going to stderr, they are outputs from the simulation not errors. If the assertion does indicate an error condition in the simulation then it might still makes sense to send this to stdout, it is a simulation error not a simulator error.

tgingold, 2016-12-12:
Yes, report and assert strings go to stderr.

I plan to close this issue as there is nothing to do.

Paebbels, 2016-12-12:
@markNZed Maybe you are interested in the [`--unbuffered`](http://ghdl.readthedocs.io/en/latest/Simulation_and_runtime.html?highlight=buffered#cmdoption--unbuffered) command line switch.

markNZed, 2016-12-13:
@Paebbels yes, I've started using that thanks. It shows up as an option of the executable but is not showing as an option of ghdl - took a while to track it down.

henrikbrixandersen, 2016-12-13:
@markNZed Another option, which I employ, is to direct simulator output to a log file from within the VHDL code itself.

OSVVM (https://github.com/OSVVM/OSVVM/) provides a nice API for this functionality in its TranscriptPkg and AlertLogPkg packages.

Paebbels, 2016-12-13:
Similar logging features can be found in frameworks like [UVVM](https://github.com/UVVM/UVVM_All?ts=2) or [VUnit](https://github.com/VUnit/vunit).

closed 2016-12-17 by tgingold