ghdl/ghdl #224: --read-wave-opt first element of the path not found in design

markNZed, 2016-12-12:
I'm using --write-wave-opt=wave-opt-out.txt then modifying that file and using it with --read-wave-opt and get the following warning:

warning: wave-opt-in.txt:70:5: \u_ctr : first element of the path not found in design. More references may follow

The VCD does not include many of the signals, the signal which generated the warning is:

/tb/\U_CTR/reg_pc_15_11[4]\

Notice the escaping, so \U_CTR/reg_pc_15_11[4]\ is a signal name, it is not indicating hierarchy.

Paebbels, 2016-12-12:
According to @Jonsba, extended identifiers are not yet supported.

Jonsba, 2016-12-12:
I guess that, unless not supported by GHDL, it will still work with 
signal selection as long as they don't have a / in them (or a . for 
signal paths in packages).

markNZed, 2016-12-12:
Unfortunately the design I have does have the '/' character in the extended identifier.

I don't know ADA but here is what seems to be working:

In grt-wave_opt-file.adb procedure Parse_Path add a variable at line 239

`ExtendedName : Boolean := False; `

At line 270 inside the "Find next identifier" loop replace line

`if Line (Last) = Seps (Tree_Index) then`

With:

```
            if Line (Last) = '\' then
               ExtendedName := not ExtendedName;
               Report_E("ExtendedName set");
            end if;
            if not ExtendedName and Line (Last) = Seps (Tree_Index) then

```

markNZed, 2016-12-12:
This highlighted another issue in the design I have, VHDL is case sensitive for extended identifiers and line 241 of  grt-wave_opt-file.adb is: 

To_Lower (Line);

I can probably work around it by converting the design. But it is probably something to note. The comments in that file are misleading "This is fine now, but maybe on day Verilog and/or System-C will be  supported by GHDL and they are case-sensitive languages."

Jonsba, 2016-12-12:
Ok I will have time to look on that this week-end. Thanks for the 
suggested hints. After the corrections, the lowering of the line will 
only lower identifiers that aren't extended.

Le 12. 12. 16 à 20:52, markNZed a écrit :

Jonsba, 2016-12-18:
Hello,

I have added support for extended identifiers in wave option file. This 
also fixes the issue with case sensitiveness. Could you please clone my 
repository (git clone -b jonsba/signals_selection 
https://github.com/Jonsba/ghdl) and try if it works as expected ? If 
everything is fine, I will then make a pull request.

Have a nice evening !
Jonas

tgingold, 2016-12-18:
On 18/12/16 18:36, Jonas Baggett wrote:
> Hello,
>
> I have added support for extended identifiers in wave option file. This
> also fixes the issue with case sensitiveness. Could you please clone my
> repository (git clone -b [...]

If you have only changed files in grt, you can simply run ./testsuite.sh
in testsuite/gna
There are no other tests, so if it is OK, you can do a pull request.
(Do not hesitate to add tests).

Thanks,
Tristan.

Jonsba, 2016-12-19:
Ok, I did the pull request but the checks with appveyor failed with the 
following message : "Specify a project or solution file. The directory 
does not contain a project or solution file.". It seems it isn't 
correctly configured yet.

Paebbels, 2016-12-19:
@Jonsba The AppVeyor CI flow is new and enabled for all branches. It misses its config file.

This might be a workaround:
```xml
build: off
```

@tgingold Please enabled the AppVeyor setting: Build only if `.appveyor.yml` exists.

tgingold, 2016-12-20:
On 19/12/16 20:09, Patrick Lehmann wrote:
> @Jonsba <https://github.com/Jonsba> The AppVeyor CI flow is new and
> enabled for all branches. It misses its config file.
>
> This might be a workaround:
>
> build: off
>
> @tgingold <https://github.com/tgingold> Please enabled the AppVeyor
> setting: Build only if |.appveyor.yml| exists.

I have ticked the setting: Skip branches without appveyor.yml

Tristan.

Jonsba, 2016-12-20:
Thanks, now the checks have passed.

closed 2017-01-13 by markNZed