ghdl/ghdl #225: regression, function call in port association

thomashk0, 2016-12-14:
The following code compiles nicely on branch 2016-09-14 but causes GHDL to crash on master:

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity foo is
  port (
    a0 : in unsigned(1 downto 0)
    );
end entity;

architecture bar of foo is
begin
  assert a0(0) = '0';
end architecture;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity foo_tb is
  generic
  ( DEFAULT_X : unsigned(1 downto 0) := (others => '0')
  );
end entity;

architecture tb of foo_tb is

  function compute_stuff_with_x(x : unsigned) return unsigned is
  begin
    return x + 1;
  end compute_stuff_with_x;

begin

  foo_inst:
    entity work.foo
    port map
    ( a0 => compute_stuff_with_x(DEFAULT_X)
    );

end architecture;
```

It doen't seem backend related...Here is the output:

```
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (2016-09-14-223-g9c5f5e3) [Dunoon edition]
Compiled with GNAT Version: 6.2.1 20160830
In directory: /home/tom/code/hardware-design/ghdl-bugs/unsigned-bound-check-failure/reproduce/
Command line:
/home/tom/local/GHDL/GHDL_install/libexec/gcc/x86_64-unknown-linux-gnu/4.9.3/ghdl1 -P/home/tom/local/GHDL/GHDL_install/lib/ghdl/v93/std/ -P/home/tom/local/GHDL/GHDL_install/lib/ghdl/v93/ieee/ -quiet -o foo_tb.s foo_tb.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : iirs.adb:1909
Call stack traceback locations:
0x100ad5f 0x64092d 0x6ded23 0x6af7cd 0x6bbd9b 0x6bd08c 0x6d144e 0x6d2114 0x6fd719 0x6c9b7a 0x6ff292 0x6496ed 0x6244ce 0xa016d0 0xa032be 0x649749 0x6216f5 0x7f20380f828f 0x622068 0xfffffffffffffffe
******************************************************************
```

The full reproducer:

```bash

#!/bin/bash

cat > foo.vhd << EOF
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity foo is
  port (
    a0 : in unsigned(1 downto 0)
    );
end entity;

architecture bar of foo is
begin
  assert a0(0) = '0';
end architecture;
EOF

cat > foo_tb.vhd << EOF
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity foo_tb is
  generic
  ( DEFAULT_X : unsigned(1 downto 0) := (others => '0')
  );
end entity;

architecture tb of foo_tb is

  function compute_stuff_with_x(x : unsigned) return unsigned is
  begin
    return x + 1;
  end compute_stuff_with_x;

begin

  foo_inst:
    entity work.foo
    port map
    ( a0 => compute_stuff_with_x(DEFAULT_X)
    );

end architecture;
EOF

ghdl -i *.vhd
ghdl -m foo_tb
```

Any idea on this ?

tgingold, 2016-12-15:
On 14/12/16 21:39, thomashk0 wrote:
> The following code compiles nicely on branch 2016-09-14 but causes GHDL
> to crash on master:

I can reproduce the crash and I am investigating.

thomashk0, 2016-12-15:
Alright, thanks !

Meanwhile the crash can be avoided by using an intermediate constant to store the actual function call result.

commit 60f245f0 (2016-12-16): assocs: rework conversion.

An actual is analyzed as a conversion if the inner expression is an object of
the right class.

Fix #225
closed 2016-12-16 by tgingold