ghdl/ghdl #227: Error: Not static choise exclude others choise

ljjterma, 2016-12-15:
The following code works fine in Precision, Vivado and Questasim, but produces an error in gdhl (probably a VHDL 2008 issue).

not static choise exclude others choise
non-locally static choise for an aggregate is allowed only if only choise

```VHDL

library ieee;
use ieee.std_logic_1164.all;
use ieee.std_logic_arith.all;

entity test is
end entity test;

architecture rtl of test is

  type test_t is record
    t1 : natural;
    t2 : natural;
  end record test_t;

  constant C_TEST_T : test_t :=
    (
      t1 => 1,
      t2 => 2);

  constant C_TEST : std_ulogic_vector(0 to 7) :=
    (C_TEST_T.t1 => '1', others => '0');

begin

end architecture rtl;

```

tgingold, 2016-12-15:
On 15/12/16 09:01, ljjterma wrote:
> The following code works fine in Precision, Vivado and Questasim, but
> produces an error in gdhl (probably a VHDL 2008 issue).
>
> not static choise exclude others choise
> non-locally static choise for an aggregate is allowed only if only choise

Technically, this is the same as issue 216 (your favorite one!): in vhdl 
2008, C_test_t.t1 is locally static.

Tristan.

Paebbels, 2016-12-15:
I don't think it's allowed in VHDL anyway, because there is a proposal at the IEEE working group discussing if the restriction can be lifted: [Relaxed Others Aggregate](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/RelaxedOthersAggregate).

tgingold, 2016-12-15:
On 15/12/16 09:06, Patrick Lehmann wrote:
> I don't think it's allowed in VHDL anyway, because there is a proposal
> at the IEEE working group discussing if the restriction can be lifted:
> Relaxed Others Aggregate
> <http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/RelaxedOthersAggregate>.

I don't agree.  This issue is not really related to aggregates.
 From the aggregate POV, this is not different from:
   (1 => '1', others => '0')
which is certainly valid.

ljjterma, 2016-12-15:
I thought it looked a bit different than 216, so i posted the error. Sorry, my mistake. The issues i have posted so far seems to be the only compile issues i have.

tgingold, 2016-12-15:
On 15/12/16 09:22, ljjterma wrote:
> I thought it looked a bit different than 216, so i posted the error.

No problem with that.  It will become a testcase!

> Sorry, my mistake. The issues i have posted so far seems to be the only
> compile issues i have.

Great!

diogratia, 2016-12-18:
It's easy to get misled as to what's going on here. We have an element of a record constant initialized by a record aggregate used as a choice in an array aggregate.

**First let's look at why this fails with earlier revisions of VHDL:**

```
ghdl -a test.vhdl
test.vhdl:21:5: not static choice exclude others choice
test.vhdl:21:5: non-locally static choice for an aggregate is allowed only if only choice
ghdl: compilation error
```
ghdl defaults to --std=93c (-1993 with some relaxed rules in the face of unpublished interpretations of the standard).

And it becomes a little better explained in another tool:
```
nvc -a test.vhdl
** Error: non-locally static choice must be only choice
	File test.vhdl, Line 21
        (C_TEST_T.t1 => '1', others => '0');
         ^^^^^^^^^^^
```
And this points to the paragraph in the standard.

>7.3.2.2 Array Aggregates para 2:  
>  
>Apart from a final element association with the single choice **others**, the rest (if any) of the element associations of an array aggregate must be either all positional or all named. ***A named association of an array aggregate is allowed to have a choice that is not locally static, or likewise a choice that is a null range, only if the aggregate includes a single element association and this element association has a single choice.*** An **others** choice is locally static if the applicable index constraint is locally static.

The issue is the choice C_TEST.t1 is not the only choice or it isn't locally static.

>6.1 Names para 8 (in part):  
>  
>Furthermore, a name is said to be a locally static name if and only if one of the following conditions hold:

>>    -- The name is a simple name or selected name (including those that are expanded names) that is not an alias and that does not denote a function call or an object or a value of an access type and (in the case of a selected name) whose prefix is a locally static name. 

So is selected name C_TEST_T.t locally static?  The value of a variable or signal is never locally static but C_TEST_T is a constant:
```vhdl
  constant C_TEST_T : test_t :=
    (
      t1 => 1,
      t2 => 2);
 ```     
The values assigned to the elements of C_TEST_T are locally static -

>7.4.1 Locally static primaries, para 1 (in part):  
>  
> An expression is said to be locally static if and only if every operator in the expression denotes an implicitly defined operator whose operands and result are scalar and if every primary in the expression is a locally static primary, where a locally static primary is defined to be one of the following:

>>   a. A literal of any type other than type TIME   

>>   b. A constant (other than a deferred constant) explicitly declared by a constant declaration and initialized with a locally static expression 

So what's stopping C_TEST_T.t1 from being locally static?

If we look in

7.4.2 Globally static primaries para 1 (in part):

>7.4.2 Globally static primaries  
>  
>An expression is said to be globally static if and only if every operator in the expression denotes a pure function and every primary in the expression is a globally static primary, where a globally static primary is a primary that, if it denotes an object or a function, does not denote a dynamically elaborated named entity (see 12.5 ) and is one of the following:  
>> ...  
>> b. A locally static primary  
>> ...  
>> e. A constant (including a deferred constant)  
>> ...   
>> h. A record aggregate, if and only if all expressions in its element associations are globally static expressions 

We find that a record aggregate is only globally static (and we see a literal of other than type TIME is locally static from 7.4.1 a.).

The aggregate used to initialized the constant is globally static but not locally static.

**So What changed in -2008?**

Other than 7.4 (mapping to 9.4 in -2008), The paragraphs from 6.1 (8.1 -2008) and 7.3.2.2 (9.3.3.3 -2008) remain the same.

What's different in -2008 is the aggregate providing the initial value to the C_TEST constant is now locally constant:

>9.4.2 Locally static primaries  
>  
>An expression is said to be locally static if and only if every operator in the expression denotes an implicitly defined operator or an operator defined in one of the packages STD_LOGIC_1164, NUMERIC_BIT, NUMERIC_STD, NUMERIC_BIT_UNSIGNED, or NUMERIC_STD_UNSIGNED in library IEEE, and if every primary in the expression is a *locally static primary*, where a locally static primary is defined to be one of the following:
>>...  
>> m) A record aggregate in which all expressions in element associations are locally static expressions

And that makes the use of choice C_TEST_T.t1 legal for -2008 in the initialization of constant C_TEST:

```vhdl
constant C_TEST : std_ulogic_vector(0 to 7) :=
  (C_TEST_T.t1 => '1', others => '0');
```
It's now locally static allowing the use of a second choice (**others**).

commit fb514835 (2016-12-20): vhdl2008: evaluate static record aggregate.

Fix #227
closed 2016-12-20 by tgingold