ghdl/ghdl #228: remove note for PSL coverage ?

thomashk0, 2016-12-15:
Hi,

Currently each time a PSL coverage statement is hit, the GHDL run-time reports a note about it (see [simulation-main.adb](https://github.com/tgingold/ghdl/blob/master/src/vhdl/simulate/simulation-main.adb#L447-L449)). It makes the output quite unreadable :(

Can I remove this report ?
Or is it better to introduce a command line option like ```--assert-verbosity={note,info,warning,error,failure}``` ?

tmeissner, 2016-12-15:
I use PSL coverage sometimes. For me, it would be sufficient, if there is a message only when the cover directive is used together with the report statement like this:

`test_c : cover {a = b} report "test_c covered";`

There don't have to be a message when it is used without the report statement:

`test_c : cover {a = b};`

I think both versions are counted and you get the counts with the --psl_report option after simulation. So there would be no loss of information.

tgingold, 2016-12-16:
On 15/12/16 18:53, thomashk0 wrote:
> Hi,
>
> Currently each time a PSL coverage statement is hit, the GHDL run-time
> reports a note about it (see simulation-main.adb [...]

Would you be happy with --psl-report=no ?

thomashk0, 2016-12-16:
@tgingold I'm affraid this option would prevent the generation of JSON reports

The proposition of tmeissner seems pretty good.

PR #251 (2017-01-06): removing PSL coverage hit default report
It closes #228.

Nothing is displayed unless the user specifies a report statement.
Testcase added to check this behaviour.
commit 46b02a2f (2017-01-06): removing PSL coverage hit default report

It closes #228.

Nothing is displayed unless the user specifies a report statement.
Testcase added to check this behaviour.
closed 2017-01-06 by tgingold