ghdl/ghdl #229: VHDL 2008 and cocotb?

javValverde, 2016-12-16:
Hello,

I am able to run the adder example of cocotb with ghdl as a simulator successfully.

But when I try to use VHDL 2008 (i.e. using `--std=08`, and the adder vhdl file unchanged), I get the following error:
```
/usr/local/bin/ghdl -r adder --vpi=/cocotb/build/libs/x86_64/libvpi.so 
/usr/local/bin/ghdl: cannot find entity or configuration adder
```

I modify the Makefile.ghdl in cocotb so that it uses the `--std=08` flag. The changes look like this:
```
analyse: $(VHDL_SOURCES) $(SIM_BUILD)
        cd $(SIM_BUILD) && $(CMD) -a --std=08 --work=$(RTL_LIBRARY) $(VHDL_SOURCES) && $(CMD) -e --std=08 --work=$(RTL_LIBRARY) $(TOPLEVEL)
```

If I use any other vhdl version (e.g `--std=00`), it works perfectly. 

BTW, the git versions are:
- ghdl, `9a0c7b4a7e7850187a81f1bf8c141a2567efe543`
- cocotb `07c079068b45fb65400bb8451da7844150d7ec1d`

What am I doing wrong???

Paebbels, 2016-12-16:
You have to apply `--std=08` also to the run command (`-r`). Each VHDL revision creates distinct files on the disk. `--std=08` is required to select the correct files (not the VHDL-93 files).

closed 2016-12-17 by tgingold

javValverde, 2016-12-19:
Hi guys,

I just tested again using the `--std=08` for the run command and it is also not woking...

The error is:
```
/usr/local/bin/ghdl -r adder --std=08 --vpi=/cocotb/build/libs/x86_64/libvpi.so 
/usr/local/bin/ghdl: cannot find entity or configuration adder
```

The Makefile.ghdl is:
```
# Compilation phase
analyse: $(VHDL_SOURCES) $(SIM_BUILD)
        cd $(SIM_BUILD) && $(CMD) -a --std=08 --work=$(RTL_LIBRARY) $(VHDL_SOURCES) && $(CMD) -e --std=08 --work=$(RTL_LIBRARY) $(TOPLEVEL)

results.xml: analyse  $(COCOTB_LIBS) $(COCOTB_VPI_LIB)
        cd $(SIM_BUILD); \
        PYTHONPATH=$(LIB_DIR):$(SIM_ROOT):$(PWD):$(PYTHONPATH) LD_LIBRARY_PATH=$(LIB_DIR):$(LD_LIBRARY_PATH) MODULE=$(MODULE) \
        TESTCASE=$(TESTCASE) TOPLEVEL=$(TOPLEVEL) TOPLEVEL_LANG=$(TOPLEVEL_LANG) \
        $(CMD) -r $(TOPLEVEL) --std=08 --vpi=$(LIB_DIR)/libvpi.$(LIB_EXT) $(SIM_ARGS)
```

Is the Makefile correct?

tgingold, 2016-12-19:
On 19/12/16 09:28, javValverde wrote:
> Hi guys,
>
> I just tested again using the |--std=08| for the run command and it is
> also not woking...
>
> The error is:
>
> |/usr/local/bin/ghdl -r adder --std=08
> --vpi=/cocotb/build/libs/x86_64/libvpi.so /usr/local/bin/ghdl: cannot
> find entity or configuration adder |

Try: ghdl -r --std=08 adder --vpi=...

Tristan.

javValverde, 2016-12-19:
That worked, thanks...

I'm thinking of making a pull request to the cocotb repository, adding the `--std=08`. Is there any reasons why `--std=08` should not be the default?

Paebbels, 2016-12-19:
Be aware that Cocotb it's very active since one year :(.