ghdl/ghdl #232: Error: NULL access dereferenced

ljjterma, 2016-12-19:
Simulating the code below returns the error (the code works fine in Questasim):

ghdl:error: NULL access dereferenced
  from: process test.test(rtl).P0 at test_pkg_rtl.vhd:11
ghdl:error: simulation failed

```VHDL

ENTITY test IS
END ENTITY test;

ARCHITECTURE rtl OF test IS

  SIGNAL t1, t2 : time;

BEGIN

  t2 <= abs(t1 - 10 ns);

END ARCHITECTURE rtl;

```

tgingold, 2016-12-19:
On 19/12/16 10:47, ljjterma wrote:
> Simulating the code below returns the error (the code works fine in
> Questasim):
>
> ghdl:error: NULL access dereferenced
> from: process test.test(rtl).P0 at test_pkg_rtl.vhd:11
> ghdl:error: simulation failed
>
> ENTITY test IS
> END ENTITY test;
>
> ARCHITECTURE rtl OF test IS
>
>   SIGNAL t1, t2 : time;
>
> BEGIN
>
>   t2 <= abs(t1 - 10 ns);
>
> END ARCHITECTURE rtl;

The error message is certainly misleading, but the vhdl code is not 
correct.  t1 - 10 ns results in an overflow.

Are you on windows or on linux ?

ljjterma, 2016-12-19:
I'm on Linux

tgingold, 2016-12-19:
On 19/12/16 13:00, ljjterma wrote:
> I'm on Linux

Ok.  Which back-end are you using, llvm or mcode ?

ljjterma, 2016-12-19:
mcode

tgingold, 2016-12-19:
On 19/12/16 13:14, ljjterma wrote:
> mcode

Ok.  Is it a blocking issue for you ?
Can you fix this issue in your sources ?

Tristan.

ljjterma, 2016-12-19:
It is not a blocking issue. Issue #216 is still my favorite.

tgingold, 2016-12-19:
On 19/12/16 13:43, ljjterma wrote:
> It is not a blocking issue. Issue #216
> <https://github.com/tgingold/ghdl/issues/216> is still my favorite.

I have worked on it and I almost have implemented the feature for #216.

ljjterma, 2016-12-19:
Sweet!

closed 2016-12-24 by tgingold