ghdl/ghdl #234: Error: Attribute on port

ljjterma, 2016-12-19:
The code below generates the following error (the code works fine in Questasim, Vivado and Precision)

no named entities "o" in declarative part

```VHDL

ENTITY test IS
  PORT (
    o : OUT integer);
END ENTITY test;

ARCHITECTURE rtl OF test IS

  ATTRIBUTE block_rom      : boolean;
  ATTRIBUTE block_rom OF o : SIGNAL IS true;
  
BEGIN

END ARCHITECTURE rtl;

```

Paebbels, 2016-12-19:
I think GHDL is correct. Attributes must be specified in the same declarative region as the object they should decorate. So port related attributes must be declared in the entity not in the architecture.

**Test this:**
```
ENTITY test IS
  PORT (
    o : OUT integer
  );
  ATTRIBUTE block_rom      : boolean;
  ATTRIBUTE block_rom OF o : SIGNAL IS true;
END ENTITY test;
```

**LRM §7.2 Attribute Specification:**
> An attribute specification for an attribute of an entity declaration, an architecture, a configuration, or a
> package shall appear immediately within the declarative part of that declaration.

tgingold, 2016-12-20:
On 19/12/16 17:45, Patrick Lehmann wrote:
> I think GHDL is correct. Attributes must be specified in the same
> declarative region as the object they should decorate. So port related
> attributes must be declared in the entity not in the [...]

Yes, GHDL is correct here.

ljjterma, 2016-12-20:
Putting the attribute in the entity declaration works fine in Questasim, so I am happy.

closed 2016-12-20 by ljjterma