ghdl/ghdl #235: Internal error

ljjterma, 2016-12-19:
When simulating parts of my code I get the error below. I cannot send the code, but is there any way I can narrow it down (display the file and line number), so that I can make a test case? Or is the error message enough for you?

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: GPL 2016 (20160515-49)
In directory: /home/ljj/work_SBL-1012/hw/hds/application/astra/buildtools/
Command line:
ghdl -r -g --work=math --std=08 --workdir=/opt/ljj/work_SBL-1012/sim/math/v08 --ieee=synopsys -P/opt/ljj/work_SBL-1012/sim/ -P/opt/ljj/work_SBL-1012/gcl/xilinx-ise unit_tests --stop-time=10ms
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap7.adb:4531
Call stack traceback locations:
0x5bb6bb 0x5bbd77 0x5bf061 0x60e1a5 0x623ed5 0x624331 0x5d964e 0x62a2bf 0x5e35de 0x62e9cd 0x635a47 0x6347da 0x4cd82a 0x4053d7 0x7f63094afb13 0x404257 0xfffffffffffffffe
******************************************************************

tgingold, 2016-12-19:
On 19/12/16 12:19, ljjterma wrote:
> When simulating parts of my code I get the error below. I cannot send
> the code, but is there any way I can narrow it down (display the file
> and line number), so that I can make a test case? Or is the error
> message enough for you?

I only have a rough idea of the issue.

Can you try to enable the verbose mode:  ghdl -r -v -v ...
It will display which units are analyzed, so you will know the
unit in which the error occurs.

You can also replace the 'raise Internal_Error;' statement at
line trans-chap7.adb:4531 by:
Error_Kind ("Translate_Equality", Etype);
That might help.

Tristan.

ljjterma, 2016-12-20:
Using the -v option and backtracing in gdb, I narrow it down to the code below. I guess, the problem is probably that VHDL 2008 unconstrained arrays are not supported yet.

```VHDL

PACKAGE test_pkg IS

  TYPE string_array_t IS ARRAY (natural RANGE <>) OF string;

END test_pkg;

ENTITY test IS
END ENTITY test;

LIBRARY work;
USE work.test_pkg.string_array_t;

ARCHITECTURE rtl OF test IS
  

BEGIN

END ARCHITECTURE rtl;

```

ljjterma, 2016-12-20:
I noticed that the issue is a duplicate of #238

QuantumRipple, 2017-01-05:
I'd also like to see support for unconstrained array types of unconstrained elements. This is the killer feature of VHDL 2008 for me.

I think the only simulator I have access to that supports this is Riviera-PRO, but I prefer to use FOSS or at least free software for my non-work projects.

Paebbels, 2017-01-05:
According to the latest commit messages like "translate: WIP - refactoring for unbounded records.", Tristan is working on that feature. I'm also very excited to test it :).

tgingold, 2017-10-22:
Has been implemented.

closed 2017-10-22 by tgingold

LogicSolutionsBrundler, 2018-05-30:
The issue does not yet seem to be fully resolved. I still get:

`/psi_tb_textfile_pkg.vhd:35:31: array element of unconstrained array type "std_logic_vector" is not allowed`

For the VHDL code:

`type TextfileData_t is array (natural range <>) of std_logic_vector;`

I am using GHDL 0.35

Paebbels, 2018-05-30:
What command line did you use to call GHDL?

LogicSolutionsBrundler, 2018-06-01:
Sorry, this was my fault. I simply forgot the "--std=08" switch for elaboration. Adding the switch solved the problem.