ghdl/ghdl #238: GHDL bug on unconstrained records

henrikbrixandersen, 2016-12-19:
The following code triggers a bug in GHDL:

```vhdl
LIBRARY ieee;
USE ieee.std_logic_1164.ALL;
USE ieee.numeric_std.ALL;

ENTITY test IS
  TYPE foo_t IS RECORD
    bar : unsigned;
  END RECORD foo_t;
END ENTITY test;

ARCHITECTURE bar OF test IS
  SIGNAL baz : foo_t(bar(1 DOWNTO 0));
BEGIN

END ARCHITECTURE bar;
```

```
$ ghdl -a -v --std=08 test.vhd
$ ghdl -e -v --std=08 test

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (2016-09-14-227-g8616070) [Dunoon edition]
Compiled with GNAT Version: 4.8.5 20150623 (Red Hat 4.8.5-4)
In directory: /home/brix/tmp/
Command line:
ghdl -e -v --std=08 test
Exception TYPES.INTERNAL_ERROR raised
Exception information:
Exception name: TYPES.INTERNAL_ERROR
Message: trans-chap7.adb:4531
Call stack traceback locations:
0x5cfbdb 0x5d07a4 0x5d34e9 0x628cb9 0x60308a 0x6469d0 0x5f284b 0x645e08 0x5fcdd2 0x64a567 0x653bf9 0x650c80 0x4dcc64 0x409946 0x7f0cd2b5ab13 0x408777 0xfffffffffffffffe
******************************************************************
```

Paebbels, 2016-12-19:
Unconstrained arrays in records are not yet supported by GHDL in VHDL-2008.

-------------------
Related issues:
* #195

henrikbrixandersen, 2016-12-20:
Do you maintain a list of currently supported VHDL-2008 features (or perhaps even better; a list of features not yet supported)?

henrikbrixandersen, 2016-12-20:
A variant of this was also reported as #70.

Paebbels, 2016-12-20:
I don't know if @tgingold has an overview list on working / missing features for VHDL-2008. But as the LCS phase has started for VHDL-2017, we could create a list for the next VHDL release :).

tgingold, 2016-12-21:
On 20/12/16 17:05, Henrik Brix Andersen wrote:
> Do you maintain a list of currently supported VHDL-2008 features (or
> perhaps even better; a list of features not yet supported)?

Not really.  This is what I have in my notes:

* Unconstrained Element Types
* DONE: Relational operators: ??, ?=, ?/=, ?>, ?>=, ?<, ?<=
* DONE: Maximum and Minimum
* DONE: shift operators
* DONE: bit_vector, boolean_vector, integer_vector and the predefined 
operators
* ???: Strength Reduction Functions
* DONE: Unary reduction Operators and,or,nand,nor,xor,xnor
* DONE: Array Logic Operators
* DONE: Scalar Logic Operators
* DONE: If-else-if & Case Generate
* Sequential Signal Assignments
* Matching Select for Variables
* Matching Case for Variables
* Matching Select for Signals
* Matching Case for Signals
* Case? Statement
* Select? Statement
* Slices in aggregates
* DONE: Sized Bit String Literals
* DONE: Reading Output Ports
* Expressions in Port Maps
* DONE: Process(all)
* DONE: Referencing Generics in Generic Lists
* Relaxed return rules for Function Return Values
* Relaxed Qualified Expressions
* Type Conversions
* Extensions to globally static and locally static expressions
* Static Ranges and Integer expressions in range bounds
* DONE: Block Comments
* DONE: std_logic_1164/Numeric_bit/Numeric_std

henrikbrixandersen, 2016-12-21:
Excellent, thank you. Keep up the good work!

closed 2017-06-18 by tgingold