ghdl/ghdl #243: can't match character literal '0' with type std_ulogic

ljjterma, 2016-12-21:
The code below works fine in Questasim, Precision and Vivado, but ghdl reports the errors:

-*- mode: compilation; default-directory: "/opt/ljj/work_SBL-1012/hw/hdl/test/" -*-
Compilation started at Wed Dec 21 09:59:53

ghdl -a --std=08 --workdir=./ -P$WORK/sim --ieee=synopsys -fexplicit  test1.vhd
test1.vhd:18:37:warning: suspicious parenthesis around aggregate
test1.vhd:17:58: can't match character literal '0' with type std_ulogic
test1.vhd:18:37: 'others' choice not allowed for an aggregate in this context

Compilation exited abnormally with code 1 at Wed Dec 21 09:59:53

```VHDL

LIBRARY ieee;
USE ieee.std_logic_1164.ALL;

PACKAGE test_pkg IS

  SUBTYPE test_t IS std_ulogic_vector(7 DOWNTO 0);

  TYPE test_array_t IS ARRAY (natural RANGE <>) OF test_t;

END PACKAGE test_pkg;

LIBRARY work;
USE work.test_pkg.ALL;

ENTITY test IS
  PORT (
    a : IN test_array_t(0 TO 4) := (OTHERS => (OTHERS => '0'));
    b : IN test_array_t(0 TO 4) := ((OTHERS => (OTHERS => '0'))));
END ENTITY test;

ARCHITECTURE rtl OF test IS

BEGIN

END ARCHITECTURE rtl;

```

tgingold, 2016-12-21:
I think ghdl is correct for both cases.
Line 17: the '0' of std_ulogic is not visible, you need to add 'use ieee.std_logic_1164.all;' before the entity declaration.
Line 18: you cannot use 'others =>' in that context. See LRM08 9.3.3.3 for details if needed.

Tristan.

tgingold, 2016-12-22:
The error message is now:

test.vhdl:17:58: enumeration literal "'0'" is not visible (add a use clause)

larsjoost, 2016-12-22:
The code is a boil down of production code, that fails when being compiled by ghdl but works fine in Questasim, Precision and Vivado. I guess I messed up the boiling part. I will try to see if I can make some new test code that shows the error.

tgingold, 2016-12-22:
On 22/12/16 09:05, larsjoost wrote:
> The code is a boil down of production code, that fails when being
> compiled by ghdl but works fine in Questasim, Precision and Vivado. I
> guess I messed up the boiling part. I will try to see if I can make some
> new test code that shows the error.

Note there are two different errors: on line 17 and on line 18.

Ghdl follows very closely the standard and is therefore very strict.

Tristan.

closed 2016-12-24 by tgingold