ghdl/ghdl #246: Winbuild.ps1: Installing GHDL into an existing directory gives an error  while writing InstallDir.conf.

Paebbels, 2016-12-30:
Installing GHDL with `winbuild.ps1 -Install <Path>` results in an error while writing `InstallDir.conf`:

```
D:\git\GitHub\ghdl\dist\mcode [paebbels/master = +2 ~0 -0 !]> .\winbuild.ps1 -Install C:\Tools\GHDL\0.34-dev-gnatgpl32-mcode
================================================================================
GHDL for Windows - GHDL compile and bundle script
================================================================================
Installing GHDL 0.34-dev for Windows...
  Writing installation path to 'build\InstallDir.conf'...
Out-File : Ein Teil des Pfades "D:\git\GitHub\ghdl\dist\mcode\build\InstallDir.conf" konnte nicht gefunden werden.
At D:\git\GitHub\ghdl\dist\mcode\winbuild.ps1:526 char:19
+ ...     $InstallPath | Out-File -FilePath $InstallDirFile -Encoding Ascii
+                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : OpenError: (:) [Out-File], DirectoryNotFoundException
    + FullyQualifiedErrorId : FileOpenFailure,Microsoft.PowerShell.Commands.OutFileCommand

[ERROR]: Directory 'C:\Tools\GHDL\0.34-dev-gnatgpl32-mcode' already exists.
```

dalex78, 2026-08-17:
# Issue #246 — Winbuild.ps1: installing into an existing directory fails writing InstallDir.conf

https://github.com/ghdl/ghdl/issues/246

Analysis by an AI agent. **Recommend closing the issue**

## High-level summary

The report is against `winbuild.ps1`, a PowerShell script that used to drive GHDL's native Windows (MSVC + GNAT) build and install it with `winbuild.ps1 -Install <Path>`. Installing into a path that already exists made the script fail while writing `build\InstallDir.conf` (`DirectoryNotFoundException` from `Out-File`), immediately followed by the script's own `[ERROR]: Directory '...' already exists.` check.

## Investigation

`winbuild.ps1` no longer exists anywhere in the current `master` tree (only `scripts/windows/mcode/winbuild.bat`, an unrelated legacy batch file, remains under that name). The native PowerShell+MSVC Windows build system was replaced by an MSYS2-based build (see `scripts/windows/targets.psm1` / `scripts/windows/shared.psm1`, and the `Build-MSYS2.yml` GitHub Actions workflow, which drives the ordinary Unix-style `./configure && make` flow through MSYS2's bash). `targets.psm1` itself has no import/call site anywhere in the repository (`grep -rl Import-Module scripts/windows` finds nothing), so it is dead code, not part of any currently exercised build path.

Since the exact script this bug was reported against has been deleted and its conceptual successor (`targets.psm1`) isn't wired into any current build or CI job, there is no live code path left to reproduce this bug against, so no minimal working example can be constructed for it on current `master`.

## What the fix does

No fix is applied. This bug is obsolete on current `master`: the reported script is gone, and the Windows build now goes through MSYS2's standard `configure`/`make`, which does not have this install-path problem (it writes build state into an out-of-tree build directory the normal Unix way, not via a bespoke `InstallDir.conf` writer). **Recommend closing the upstream issue** as "fixed by build-system migration" rather than pursuing a code fix here.

tgingold, 2026-08-17:
OK!

closed 2026-08-17 by tgingold

Paebbels, 2026-08-17:
@dalex78 I think the old `winbuild.bat` can also be deleted. It's a legacy of a legacy system :)