ghdl/ghdl #247: GHDL interprets VITAL warning as error for Altera library [2008] no-vital-checks

petspats, 2017-01-02:
Quartus version: 16.1
Compile Altera library with '/compile-altera.sh -a --vhdl2008'
output : http://pastebin.com/38VJJyjm
One warning in particular will cause problems later on: 

> ../../src/vital2000/timing_p.vhdl:152:14: warning: result subtype of a pure function cannot have access subelements

Next try to use compile library with Altera netlist (post-analysis, should not contain any timing):
```
$ ghdl -a --no-vital-checks --std=08 --work=work quartus_project.vho
../../src/vital2000/timing_p.vhdl:152:14: result subtype of a pure function cannot have access subelements
$ echo $?
1

```
quartus_project.vho source: http://pastebin.com/ufVfDnYk

Current workaround is to compile Altera libraries as 93 and add `--ieee=synopsys` to command line.

doppioandante, 2018-11-30:
I'm getting the same error while trying to simulate a post-route xilinx design.
I think I can solve by rewriting my testbench as vhdl93, but is there a way to get over this error?

tgingold, 2018-11-30:
Did you try with -frelaxed-rules ?

doppioandante, 2018-11-30:
> Did you try with -frelaxed-rules ?

Thanks, this did the trick

closed 2018-12-06 by gasparka