ghdl/ghdl #250: Request: PSL 'forall' support.

lainy, 2017-01-04:
It would be very nice if the PSL **forall** operator were supported. Presently ghdl gives an error when using `-fpsl` with an assertion such as the following:
```vhdl
-- psl assert forall i in {0 to 1}: always ((a=i -> eventually! b=i));
```

The error is:
> ';' is expected instead of "i"

lainy, 2017-01-06:
@tgingold Oh jeez, I should have thought of that. Thanks so much!

lainy, 2017-01-06:
To go along with this, next_event support would be very helpful in combination with forall. However, attempting to use next_event gives the error:
> psl.sem_property: cannot handle N_NEXT_EVENT

Should I make a separate request for this?

Paebbels, 2017-01-06:
Please create a new issue in GitHub. It's better for issue tracking. You can reference this issue by using the complete URL or `#250` in you text.

lainy, 2017-01-30:
Would like to add: although generate statements do work, it's very slow for arbitrarily large sets.
For example it should be fast to use `forall` to cover all values of a 64-bit data line, since the implementation is expected to take the value symbolically, not actually instantiate 2^64 properties.

This is handy for basic sanity checking, e.g. reading from a memory returns the data that was last written might be something like:
```vhdl
-- psl assert forall faddr(31 downto 0) in boolean:
--     forall fdata(63 downto 0) in boolean:
--         always ((write='1' and addr=faddr and data=fdata) -> next_event(read)(data=fdata);
```
... which is definitely not feasible using the generate workaround.

tgingold, 2017-01-31:
I am not sure what do you mean by symbolically.

The current technology used for PSL in ghdl (NFA creation) cannot easily deal with forall.  This issue needs a rethinking.

lainy, 2017-01-31:
Ah I see. I'm not sure about the implementation details.

What I meant is that in my example property above, which has `faddr(31 downto 0)` and `fdata(63 downto 0)`, only a single property should be created, which applies to any value of `addr` and `data` as shown, rather than creating 2^64 * 2^32 individual properties, which is clearly infeasible.

I played with it more though, and I see even with `for`/`generate`, GHDL isn't creating individual properties like I thought, since it can cope with arbitrarily large numbers of `generate`'d properties - I was just confused. It is very slow however in some cases, but maybe that would be true even in the ideal case? I guess I haven't given enough thought to the implementation details...

Anyway, sorry for the confusion!

tgingold, 2017-02-06:
It's a trade-off.  You can implement a PSL property with a for all by storing the values that will be checked somewhere.  But then the issue is that you can have multiple activation in the same time (and therefore multiple values), hence the use of memory allocation.  That's fully doable, but this is not the way it is implemented in ghdl.

tmeissner, 2017-02-07:
@lainy If you try your example above in Modelsim for example, it will work. But it will run very very slow. I have similar properties in internal protocol checkers and they run very slow with it, so I mostly don't check the transferred data with such PSL forall constructs.

lainy, 2017-02-07:
@tmeissner Ahh ok, well in that case, the workaround is fine. Thanks for this information!

Topi-ab, 2025-03-08:
As a sidenote, attributes `anyconst` and `allconst` can mimc forall in some cases (in sby formal verification flow).

`assert forall i in {0 to 1}: always ((a=i -> eventually! b=i));`
=>
```
attribute anyconst: boolean;
...
signal my_i: integer range 0 to 1;
attribute anyconst of my_i: signal is true;
...
assert always a=my_i -> eventually b=my_i;
```
This triggers an assert error if any value of my_i (it checks all values) is violating the error.

`allconst` can be used in assumptions to verify that the assumption holds on all possible signal values (`anyconst` would require that at least one signal value holds the requirement). Only subset of solvers support this, and in many cases they fail to find any solution.