ghdl/ghdl #256: Legality of concurrent Assert involving 'delayed'last_event

Brian-Drummond, 2017-01-15:
Question raised on Stack Exchange, http://electronics.stackexchange.com/questions/280398/can-a-vhdl-concurrent-assertion-reference-the-last-event-attribute/280415#280415

Testcase shows an Assert (on a clock pulsewidth in a process in an entity) working, but a similar concurrent assert failing to compile, either in an entity or an architecture.

I'm not clear whether the construct is legal - the fact it's valid in a process suggests it ought to be, but I haven't studied the LRM on this - but legal or not, GHDL should handle it.

Compiler report:
ghdl -a testcase2.vhd
canon_extract_sensitivity: cannot handle IIR_KIND_LAST_EVENT_ATTRIBUTE (testcase2.vhd:4:31)

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34dev (20151126) [Dunoon edition]
Compiled with GNAT Version: 4.9.3
In directory: /home/brian/Projects/ghdl/play/
Command line:
/usr/local/libexec/gcc/x86_64-unknown-linux-gnu/4.9/ghdl1 -P/usr/local/lib/ghdl/v93/std/ -P/usr/local/lib/ghdl/v93/ieee/ -quiet -o testcase2.s testcase2.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
Exception name: TYPES.INTERNAL_ERROR
Message: errorout.adb:66
Call stack traceback locations:
0x631313 0x66a41f 0x66a350 0x66c212 0x69b2c1 0x6cb5e5 0x6227e7 0x60028e 0x9d8880 0x9da47e 0x622839 0x5fd485 0x7f7e0c9192af 0x5fdca7 0xfffffffffffffffe
******************************************************************

Execution terminated by unhandled exception
Exception name: TYPES.INTERNAL_ERROR
Message: errorout.adb:66
Call stack traceback locations:
0x631313 0x66a41f 0x66a350 0x66c212 0x69b2c1 0x6cb5e5 0x6227e7 0x60028e 0x9d8880 0x9da47e 0x622839 0x5fd485 0x7f7e0c9192af 0x5fdca7 0xfffffffffffffffe
ghdl: compilation error

[testcase.vhd.txt](https://github.com/tgingold/ghdl/files/706542/testcase.vhd.txt)
[testcase2.vhd.txt](https://github.com/tgingold/ghdl/files/706543/testcase2.vhd.txt)
[testcase3.vhd.txt](https://github.com/tgingold/ghdl/files/706544/testcase3.vhd.txt)
[testcase_testbench.vhd.txt](https://github.com/tgingold/ghdl/files/706545/testcase_testbench.vhd.txt)

Brian-Drummond, 2017-01-15:
Patch appears to work...

diff --git a/src/vhdl/canon.adb b/src/vhdl/canon.adb
index 8f23bf2d..3084d2c2 100644
--- a/src/vhdl/canon.adb
+++ b/src/vhdl/canon.adb
@@ -213,7 +213,9 @@ package body Canon is
          when Iir_Kinds_Type_Attribute =>
             null;
          when Iir_Kind_Event_Attribute
-           | Iir_Kind_Active_Attribute =>
+           | Iir_Kind_Active_Attribute
+           | Iir_Kind_Last_Event_Attribute
+           | Iir_Kind_Last_Active_Attribute =>
             --  LRM 8.1
             --  An attribute name: [...]; otherwise, apply this rule to the
             --  prefix of the attribute name.

diogratia, 2017-01-15:
As Brian notes 'last_active would also fail:
```vhdl
entity testcase2 is
    port (clk: in bit);
begin
    check: -- assert clk'delayed'last_event >= 10 ns;
    assert clk'last_active > 10 ns report "boo!";
end entity testcase2;
```

See IEEE Std 1076-2008, 8.1 para 6, clk'delayed is a *static name* and an acceptable prefix to 'LAST_EVENT denoting a static signal name (16.2.4). I tried 'LAST_ACTIVE looking for an alternative to the problem reported on Stackoverflow.

The original expression tests for a minimum clk baud interval (period) and is demonstrated in the book [The Designer's Guide to VHDL - Peter J. Ashenden - Google Books](https://books.google.com/books?id=XbZr8DurZYEC&amp;pg=PA150&amp;lpg=PA150&amp;dq=%27delayed+last_event+vhdl&amp;source=bl&amp;ots=PbdtzoGunQ&amp;sig=BxJkB1hG_tUPpWwNTLqIAedB2gU&amp;hl=en&amp;sa=X&amp;ved=0ahUKEwitjNeC1MPRAhVGFJQKHeSAAg0Q6AEIPTAG#v=onepage&amp;q=%27delayed%20last_event%20vhdl&amp;f=false "The Designer's Guide to VHDL - Peter J. Ashenden - Google Books") (Section 5.2.2 Signal Attributes). 

A case of someone trying to learn some finer points of VHDL from a book.

tgingold, 2017-01-15:
On 15/01/17 17:19, Brian-Drummond wrote:
> Patch appears to work...

I have committed a very similar patch, plus an additional
one for the first testcase.

Thank you for the issue.

Tristan.

Brian-Drummond, 2017-01-15:
Thanks Tristan ... have you pushed?

Brian-Drummond, 2017-01-15:
We may not be quite out of the woods yet
Assuming the additional patch is as follows
```
diff --git a/src/vhdl/sem_scopes.adb b/src/vhdl/sem_scopes.adb
index d6d89523..0d87c958 100644
--- a/src/vhdl/sem_scopes.adb
+++ b/src/vhdl/sem_scopes.adb
@@ -1055,7 +1055,8 @@ package body Sem_Scopes is
            | Iir_Kind_Configuration_Specification
            | Iir_Kind_Disconnection_Specification =>
             null;
-         when Iir_Kinds_Signal_Attribute =>
+         when Iir_Kinds_Signal_Attribute
+           | Iir_Kind_Signal_Attribute_Declaration =>
             null;
 
          when Iir_Kind_Protected_Type_Body =>
```
then Testcase 1 fires the assert at 30ns,, but testcases 2 and 3 fire it at 10,25 and 30ns. I can't see any reason for the behaviour to be different.

> ghdl -r testcase_testbench
> testcase.vhd:7:9:@30ns:(assertion error): Assertion violation

> ghdl -r testcase_testbench
> testcase2.vhd:4:5:@10ns:(assertion error): Assertion violation
> testcase2.vhd:4:5:@25ns:(assertion error): Assertion violation
> testcase2.vhd:4:5:@30ns:(assertion error): Assertion violation

tgingold, 2017-01-15:
On 15/01/17 18:40, Brian-Drummond wrote:
> Thanks Tristan ... have you pushed?

Not yet.  I have to run a full regression testsuite.

Tristan.

tgingold, 2017-01-15:
On 15/01/17 19:40, Brian-Drummond wrote:
> We may not be quite out of the woods yet
> Assuming the additional patch is as follows
>
> |diff --git a/src/vhdl/sem_scopes.adb b/src/vhdl/sem_scopes.adb index
> d6d89523..0d87c958 100644 --- [...]

For testcase 2 and 3, the process check is sensitive to clk'delayed.
So at 10 ns, clk'delayed is '0' at delta 0, '0' at delta 1 and '1' at 
delta 2 (so fails here).

For testcase 1, the wait of check is sensitive to clk.
So at 10 ns + 0, clk is '0' (no change.
At 10 ns + 1, clk is '1' but clk'delayed'last_event is 10 ns.
At 10ns + 2, no change.

They are not equivalent.

Tristan.

Brian-Drummond, 2017-01-15:
Thanks for the clarification. It's a subtle point, I would not have expected that clk'delayed was the implicit sensitivity list, as opposed to clk! But rewriting testcase as a sensitivity list process, and putting clk'delayed in that list does reproduce the behaviour.

Brian-Drummond, 2017-01-16:
My fault on the reproducer : `testcase_testbench` only instantiates one of the testcases at a time
 (the others didn't compile, originally).

Probably worth adding
```
dut2: entity work.testcase2(empty) port map(clk => clk);
dut3: entity work.testcase3(empty) port map(clk => clk);
```
to testcase_testbench, but I wasn't thinking ahead. Then it looks good to close.

tgingold, 2017-01-16:
On 16/01/17 19:43, Brian-Drummond wrote:
> My fault on the reproducer : |testcase_testbench| only instantiates one
> of the testcases at a time
> (the others didn't compile, originally).
>
> Probably worth adding
>
> |dut2: entity [...]

Yes, I have added a second testcase.

Brian-Drummond, 2017-01-17:
OK to close?

closed 2017-01-17 by Brian-Drummond