ghdl/ghdl #258: Different Overflow Behavior when Using Literals or Variables in Expression

preusser, 2017-01-17:
Using `GHDL 0.34-dev (2016-09-14-319-gb42069e0) [Dunoon edition]`, two arithmetically identical expressions are treated differently depending on whether one of the operands is a literal or also a variable. Is this intentional?

Test entity:
```vhdl
entity range_tb is
end range_tb;

architecture tb of range_tb is
begin
  process
    variable i : integer;
  begin

    -- Behaves like 32-bit arithmetic with modular truncation
    i := integer'low;
    report integer'image(i);
    i := integer'low-1;
    report integer'image(i);

    -- Produces the error "overflow detected"
    i := integer'low;
    report integer'image(i);
    i := i-1;
    report integer'image(i);

    wait; -- forever
  end process;
end tb;
```

Paebbels, 2017-01-17:
Possible duplicate of: [Missing bound check failure for user defined integer](https://github.com/tgingold/ghdl/issues/255)

tgingold, 2017-01-18:
Indeed, an error must be reported for line 13.

commit 097cce34 (2017-01-21): eval_is_in_bound: make it more tolerant.

Replaces check_implicit_conversion.

Fix #258
closed 2017-01-21 by tgingold