ghdl/ghdl #260: Fixed point package rounding pitfall

petspats, 2017-01-17:
This is not a bug/problem in GHDL, but i have to document my days worth of debugging somewhere..

Here is testcase that tries to convert PI into sfixed format, but gets unexpected result:
```vhdl
library ieee;
  use ieee.fixed_float_types.all;
  use ieee.fixed_pkg.all;
  use ieee.math_real.all;
use std.textio.all;

entity sfix_error is
end sfix_error;

architecture behaviour of sfix_error is
begin
   process
    variable bits: integer := 5;
    variable pi_sfix: sfixed(2 downto -bits);
    variable pi_quantized, pi_rounded, pi_back_to_real: real;
   begin
      pi_sfix := to_sfixed(3.141592653589793, 2, -bits);
      report "bits: " & to_string(pi_sfix);
      report "real: " &to_string(to_real(pi_sfix));

      -- this is the expected result
      -- quantize -> round -> 'dequantize'
      pi_quantized := 3.141592653589793 * (2.0 ** bits);
      pi_rounded := round(pi_quantized);
      pi_back_to_real := pi_rounded / (2.0 ** bits);
      report "pi_quantized: " &to_string(pi_quantized);
      report "pi_rounded: " &to_string(pi_rounded);
      report "pi_back_to_real: " &to_string(pi_back_to_real);
      wait;
   end process;
end behaviour;
```
Output (2 fractional bits):
```
$ ghdl -a --std=08 sfix_bug.vhdl
$ ghdl -r --std=08 sfix_error
sfix_bug.vhdl:21:7:@0ms:(report note): bits: 011.00
sfix_bug.vhdl:22:7:@0ms:(report note): real: 3.0                            -- real value
sfix_bug.vhdl:27:7:@0ms:(report note): pi_quantized: 1.256637061435917e1 -- 12.56
sfix_bug.vhdl:28:7:@0ms:(report note): pi_rounded: 1.3e1        -- 13
sfix_bug.vhdl:29:7:@0ms:(report note): pi_back_to_real: 3.25   -- expected value

```

Code above expects output of 3.25 but it is 3.0. Note that i use only 2 bits for fractional part, but this problem also happens for bits: 2, 5, 24, 28 (tested up to 32 bits).
Output (5 fractional bits):
```
sfix_bug.vhdl:22:7:@0ms:(report note): bits: 011.00100
sfix_bug.vhdl:23:7:@0ms:(report note): real: 3.125
sfix_bug.vhdl:30:7:@0ms:(report note): pi_quantized: 1.005309649148733e2
sfix_bug.vhdl:31:7:@0ms:(report note): pi_rounded: 1.01e2
sfix_bug.vhdl:32:7:@0ms:(report note): pi_back_to_real: 3.15625
```

It seems to me that there is some problem with the rounding algorithm. 

Workaround:
```vhdl
pi_sfix := to_sfixed(3.141592653589793, 2, -bits, guard_bits=>6);
```
For the PI case, using 6 guard bits was enough to get matching performance, tested up to 32 bits.
Note that this is float to sfix conversion function, meaning it will be done before synthesis. So basically here is an opportunity to gain precision for free.

Paebbels, 2017-01-17:
The author of the VHDL fixed and floating point packages included in the IEEE library, David Bishop (@dbishopx), has moved his repository to GitHub: https://github.com/FPHDL/fphdl. Maybe you want to ask him directly.

petspats, 2017-01-18:
I am closing this since it is not really an issue. Will do some more tests today and post it here, fphdl repo seems to be not active.

closed 2017-01-18 by petspats

dbishopx, 2017-01-18:
On 1/18/2017 3:56 AM, petspats wrote:
>
> Closed #260 <https://github.com/tgingold/ghdl/issues/260>.
>
> —
>
Can somebody tell me what this is about?
Was a bug found?

petspats, 2017-01-18:
@dbishopx 
Hey, i guess it is not an bug, just my observation that 'to_sfixed(float)' function sometimes does not yield full precision answer with default 'guard_bits'.  I ended up using 16 guard bits for this function, since it has no overhead in synthesis.

dbishopx, 2017-01-18:
On 01/18/2017 09:30 AM, petspats wrote:
>
> @dbishopx <https://github.com/dbishopx>
> Hey, i guess it is not an bug, just my observation that 
> 'to_sfixed(float)' function sometimes does not yield full precision 
> answer with default 'guard_bits'. I ended up using 16 guard bits for 
> this function, since it has no overhead in synthesis.
>
>
This will depend on the rounding mode used.
Since it uses a shift to take care of the exponent, it could shift right 
over your guard bits.