ghdl/ghdl #261: Attribute 'subtype causes exception

Paebbels, 2017-01-21:
Hello,

**The following code:**
```VHDL
entity e is
end entity;

architecture a of e is
  type AT is access INTEGER;
  type FT is file of INTEGER;
begin
  process
    variable BV : bit_vector(0 to 1);
    variable V : AT;
    file F : FT;
  begin
    report BV'simple_name;               -- BV
    report BV'subtype'simple_name;       -- _anon            [line 14]
    report BV'subtype'base'simple_name;  -- bit_vector
    report BV'element'simple_name;       -- bit
    
    report V'simple_name;                -- V
    report V'subtype'simple_name;        -- V
    report V'subtype'base'simple_name;   -- AT
    
    report F'simple_name;                -- F
    report F'subtype'simple_name;        -- FT
    report F'subtype'base'simple_name;   -- FT
    
    wait;
  end process;
end architecture;
```

**causes an exception:**
```
> ghdl.exe -a --std=08 .\access.vhdl
.\access.vhdl:14:34: 'subtype attribute is not a named entity

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (commit: 2017-01-21;  git branch: paebbels/master';  hash: 4c52dc4) [Dunoon edition]
Compiled with GNAT Version: GPL 2016 (20160515-49)
Target: i686-pc-mingw32
In directory: D:\git\GitLab\IEEE-P1076-TWiki\
Command line:
C:\Tools\GHDL\0.34-dev-gnatgpl32-mcode\bin\ghdl.exe -a --std=08 .\access.vhdl
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : nodes.adb:104 index check failed
******************************************************************
```

The expected values are taken from QuestaSim outputs.

**More Questions:**
* Why does `'subtype'simple_name` show a different behavior for access and file types?
* New exception messages sometime print an error before the crash. Is the printed error the cause of the crash or an error successfully printed and the next error causes the crash? So is line 14 or 19 related to the exception message?

Kind regards
    Patrick

--------------------------
**Related issues:**
* [Attribute 'subtype does not work](https://github.com/tgingold/ghdl/issues/253)

tgingold, 2017-01-22:
On 21/01/17 15:27, Patrick Lehmann wrote:
> Hello,
>
> *The following code:*
>
> entity e is
> end entity;
>
> architecture a of e is
>   type AT is access INTEGER;
>   type FT is file of INTEGER;
> begin
>   process
>     variable BV : bit_vector(0 to 1);
>     variable V : AT;
>     file F : FT;
>   begin
>     report BV'simple_name;               -- BV
>     report BV'subtype'simple_name;       -- _anon [line 14]
>     report BV'subtype'base'simple_name;  -- bit_vector
>     report BV'element'simple_name;       -- bit
>
>     report V'simple_name;                -- V
>     report V'subtype'simple_name;        -- V
>     report V'subtype'base'simple_name;   -- AT
>
>     report F'simple_name;                -- F
>     report F'subtype'simple_name;        -- FT
>     report F'subtype'base'simple_name;   -- FT
>
>     wait;
>   end process;
> end architecture;
>
> *causes an exception:*
>
> |> ghdl.exe -a --std=08 .\access.vhdl .\access.vhdl:14:34: 'subtype
> attribute is not a named entity ******************** GHDL Bug occured
> **************************** Please report this bug on
> https://github.com/tgingold/ghdl/issues GHDL release: GHDL 0.34-dev
> (commit: 2017-01-21; git branch: paebbels/master'; hash: 4c52dc4)
> [Dunoon edition] Compiled with GNAT Version: GPL 2016 (20160515-49)
> Target: i686-pc-mingw32 In directory: D:\git\GitLab\IEEE-P1076-TWiki\
> Command line: C:\Tools\GHDL\0.34-dev-gnatgpl32-mcode\bin\ghdl.exe -a
> --std=08 .\access.vhdl Exception CONSTRAINT_ERROR raised Exception
> information: raised CONSTRAINT_ERROR : nodes.adb:104 index check failed
> ****************************************************************** |
>
> The expected values are taken from QuestaSim outputs.
>
> *More Questions:*
>
>   * Why does |'subtype'simple_name| show a different behavior for access
>     and file types?

You should ask to Mentor !

>   * New exception messages sometime print an error before the crash. Is
>     the printed error the cause of the crash or an error successfully
>     printed and the next error causes the crash? So is line 14 or 19
>     related to the exception message?

Depends.  In that case the fallout of line 14 generated an exception.

I don't see how line 14 can be valid.  So ghdl will emit an error on it.

Currently, I have:

e1.vhdl:13:5:@0ms:(report note): bv
e1.vhdl:15:5:@0ms:(report note): bit_vector
e1.vhdl:18:5:@0ms:(report note): v
e1.vhdl:19:5:@0ms:(report note): at
e1.vhdl:20:5:@0ms:(report note): at
e1.vhdl:22:5:@0ms:(report note): f
e1.vhdl:23:5:@0ms:(report note): ft
e1.vhdl:24:5:@0ms:(report note): ft

('Element is not yet supported).

Tristan.

Paebbels, 2017-01-22:
So, your point is: The subtype is anonymous and is no named entity, thus it has no simple name, right? But how can a user know if one subtype is a named entity and another not?

LRM 8.2 lists simple names for several named entities, but it misses types and subtypes. So this brings us to the questions:
* Are types named entities?
* Why should anonymous entities have no simple name, which can be returned as a string?

Should I address such a missing attribute in another LCS?

tgingold, 2017-01-23:
On 22/01/17 21:32, Patrick Lehmann wrote:
> So, your point is: The subtype is anonymous and is no named entity, thus
> it has no simple name, right? But how can a user know if one subtype is
> a named entity and another not?

Well, the user has written the code, so he knows.

> LRM 8.2 lists simple names for several named entities, but it misses
> types and subtypes. So this brings us to the questions:
>
>   * Are types named entities?

Yes.

>   * Why should anonymous entities have no simple name, which can be
>     returned as a string?

The definition of anonymous is to have no name.

> Should I address such a missing attribute in another LCS?

Yes.

commit 5c4063d0 (2017-01-23): Allow 'subtype and 'base for 'simple_name prefix.

Fix #261
closed 2017-01-23 by tgingold