ghdl/ghdl #262: Asserts on ?= and ?/= with Don't Cares

LarsAsplund, 2017-01-21:
Expressions like these

``` vhdl
result := std_ulogic_vector'("-000") ?/= std_ulogic_vector'("0000");
result := std_ulogic_vector'("-000") ?= std_ulogic_vector'("0000");
```

cause an assertion error

``` console
std_logic_1164.vhdl:58:3:@0ms:(assertion error): STD_LOGIC_1164: '-' operand for matching ordering operator
```

?/= and ?= are no matching ordering operators so I don't think there should be an assert.

tgingold, 2017-01-24:
I see.  I think you are right.

LarsAsplund, 2017-01-24:
Any change of an update in the near future? I'm setting up a CI environment for the VHDL standardization group. This is going well. The only flaw is that a number of test cases fail due to this issue

tgingold, 2017-01-24:
On 24/01/17 12:00, Lars Asplund wrote:
> Any change of an update in the near future? I'm setting up a CI
> environment for the VHDL standardization group. This is going well. The
> only flaw is that a number of test cases fail due to this issue

Yes, the fix is now committed.

Tristan.

closed 2017-01-24 by tgingold

LarsAsplund, 2017-01-24:
Thanks Tristan for a quick response. Works like a charm!

commit a076edcd (2017-01-24): Do not emit assert error for '-' in matching equality/inequality.

Fix #262