ghdl/ghdl #264: False error message for unresolved INOUT port

Paebbels, 2017-01-28:
Hello,

this code example:
```VHDL
entity ex1_entity is
  port (
    X : inout boolean            -- std_logic works
  );
end entity;

architecture a of ex1_entity is
begin
end architecture;

library IEEE;
use     IEEE.std_logic_1164.all;

entity ex1_top is
end entity;

architecture a of ex1_top is
  signal A : std_logic;
begin
  inst : entity work.ex1_entity
    port map (
      X => A               -- line containing error
    );
end architecture;
```

Creates this error message:
> ghdl -a --std=08 .\Spaceship.vhdl
> .\Spaceship.vhdl:22:29: no interface for 'x' in association

The problem is not the missing interface in the association list. The problem is that `X` is of mode `INOUT` but not a resolved type, right?

So I think the error message is misleading.

Kind regards
    Patrick

Xiretza, 2019-11-24:
After fixing the MWE (presumably line 18 should be `signal A : boolean;` instead of `std_logic`), it seems the error message has disappeared completely:

```
$ ghdl -a ex1.vhd
$ ghdl -e ex1_top
$ ghdl -r ex1_top
```

All these commands complete successfully and produce no output. Is this the expected behaviour?

tgingold, 2019-11-25:
But the issue is about the error message, which is now:
```
ex1_top.vhdl:22:12:error: can't associate 'a' with port "x"
ex1_top.vhdl:22:12:error: (type of 'a' is std_logic)
ex1_top.vhdl:3:5:error: (type of port "x" is boolean)
```

So I think we can indeed close this issue.

closed 2019-11-25 by tgingold