ghdl/ghdl #265: Exception caused while playing with conversion functions in port maps

Paebbels, 2017-01-28:
Hello,

the following example causes an exception:

**Instance:**
```VHDL
library IEEE;
use     IEEE.std_logic_1164.all;

entity ex1_entity is
  port (
    X : inout std_logic
  );
end entity;

architecture a of ex1_entity is
begin
end architecture;
```

**Top-Level:**
```VHDL
library IEEE;
use     IEEE.std_logic_1164.all;

entity ex1_top is
end entity;

architecture a of ex1_top is
  signal A : boolean;
  
  function conv_inst_top(p : std_logic) return boolean is
  begin
    return (p = '0'); -- inverter
  end function;
  
  function conv_top_inst(p : boolean) return std_logic is
  begin
    if (p = FALSE) then
      return '0'; -- identity
    elsif (p = TRUE) then
      return '1'; -- identity
    else
      return 'X'; -- error
    end if;
  end function;
begin
  A <= TRUE;

  inst : entity work.ex1_entity
    port map (
      conv_inst_top(X) => conv_top_inst(A)
    );
end architecture;
```

The exception is:
```
> ghdl -r --std=08 ex1_top

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (commit: 2017-01-25;  git branch: paebbels/master';  hash: 7f0c5f0) [Dunoon edition]
Compiled with GNAT Version: GPL 2016 (20160515-49)
Target: i686-pc-mingw32
In directory: D:\git\GitLab\IEEE-P1076-TWiki\Spaceship\
Command line:
C:\Tools\GHDL\0.34-dev-gnatgpl32-mcode\bin\ghdl.exe -r --std=08 ex1_top
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:507
******************************************************************
```

The purpose of this example is to test if conversion functions in a port map on an INOUT port can cause a oscillation.

tgingold, 2017-01-28:
I can reproduce the issue.  It would work if you use a component instantiation.

Xiretza, 2019-11-24:
This seems to be working as expected as of 0cb817da (though the error is not as verbose as QuestaSim's):

```
$ ghdl -a --std=08 ex1_entity.vhd ex1_top.vhd
$ ghdl -e --std=08 ex1_top
$ ghdl -r --std=08 ex1_top
for signal: .ex1_top(a).a
./ex1_top:error: several sources for unresolved signal
./ex1_top:error: error during elaboration
```

Same without `--std=08`.

tgingold, 2019-11-25:
Yes, no more crashes.  At some points the error message could be improved...

closed 2019-11-25 by tgingold