ghdl/ghdl #266: Why does `-i` not return non-zero on error?

TheMassController, 2017-01-29:
The doc on the GHDL -i option writes:

`Since the files are parsed, there must be correct files. However, since they are not analyzed, many errors are tolerated by this command.`

This does not seem useful to me, in a Make environment this command will be used just before the `-r` command when a file has changed, but both will throw the same error. I would like the process to fail when the `-i` option finds an error in one of the files, so I would like for GHDL to return nonzero when `-i` fails.

Why does this command tolerate errors?

tgingold, 2017-01-31:
This command tolerates errors because it doesn't fully analyse vhdl files. And it doesn't fully analyse files so that other commands can analyse files in the correct order once all source files are known.

If you know the correct order of analysis, use directly 'ghdl -a FILES'.
If you don't know the order, you can automatically generate a makefile with 'ghdl --gen-makefile UNIT'.
If you don't care about the order, you can use 'ghdl -c FILES -r UNIT'.

Regards,
Tristan.

TheMassController, 2017-01-31:
I think I get the workflow a bit better now, thank you! One more question: why would you care about the order? Can you use `ghdl -c FILES -r UNIT` even if the order matters (this seems to be the case)?

tgingold, 2017-02-06:
You can care about the order if you have multiple architectures for one entity and rely on the default elaboration rule (use the latest architecture) for your design.

In that case, you'd better to avoid to use ghdl -c -r.
Or create a configuration.

closed 2017-02-15 by tgingold