ghdl/ghdl #267: gcov breaking

stdefeber, 2017-02-01:
Hello All,

We want to run code coverage on our design. We started of with a small IP block.
The IP is compiled with : 

ghdl -a -Wc,-ftest-coverage -Wc,-fprofile-arcs ... <unit>.vhd

The test-bench with

ghdl -a  ... tb_<unit>.vhd

GHDL shows the following error :

ghdl -e -Wl,ghdl_env_c.o --ieee=synopsys --warn-no-vital-generic --workdir=simu --work=work ax_digpin_tb                                           
simu/ax_digpin_vip.e.o: In function `_GLOBAL__sub_I_65535_0_work__ax_digpin_vip__g_day__OT__STB':
ax_digpin_vip.e.vhdl:(.text+0x5ab): undefined reference to `__gcov_init'
simu/ax_digpin_vip.e.o:(.data+0x30): undefined reference to `__gcov_merge_add'
simu/ax_digpin_clkgen.o: In function `_GLOBAL__sub_I_65535_0_work__ax_digpin_clkgen__clk_freq__OT__STB':
ax_digpin_clkgen.vhdl:(.text+0xba8): undefined reference to `__gcov_init'
simu/ax_digpin_clkgen.o:(.data+0x30): undefined reference to `__gcov_merge_add'
simu/ax_digpin_outbuf.o: In function `_GLOBAL__sub_I_65535_0_work__ax_digpin_outbuf__digpin_oe__RTI':
ax_digpin_outbuf.vhdl:(.text+0x678): undefined reference to `__gcov_init'
simu/ax_digpin_outbuf.o:(.data+0x30): undefined reference to `__gcov_merge_add'
simu/ax_digpin_ctrlreg.o: In function `_GLOBAL__sub_I_65535_0_work__ax_digpin_ctrlreg__g_day__OT__STB':
ax_digpin_ctrlreg.vhdl:(.text+0x3c6e): undefined reference to `__gcov_init'
simu/ax_digpin_ctrlreg.o:(.data+0x30): undefined reference to `__gcov_merge_add'
simu/ax_digpin_vip.a.o: In function `_GLOBAL__sub_I_65535_0_work__ax_digpin_vip__ARCH__behav__ax_digpin_clkgen__clk_freq__OT__STB':
ax_digpin_vip.a.vhdl:(.text+0x1bc5): undefined reference to `__gcov_init'
simu/ax_digpin_vip.a.o:(.data+0x30): undefined reference to `__gcov_merge_add'
collect2: error: ld returned 1 exit status
ghdl: compilation error

Any one ?

henrikbrixandersen, 2017-02-01:
Perhaps my notes on code coverage using GHDL could be useful for you?

https://blog.brixandersen.dk/2016/12/29/ghdl-gcov/

stdefeber, 2017-02-01:
Thanks a lot.
The example worked on my compiled GHDL.
I'll try to rework it for my particular IP.

best regards

ST

tgingold, 2017-02-01:
On 01/02/17 18:17, stdefeber wrote:
> Thanks a lot.
> The example worked on my compiled GHDL.
> I'll try to rework it for my particular IP.

You need to specify -ftest-coverage as a linker option when you 
elaborate the design, so that it is linked with libgcov.

Regards,
Tristan.

stdefeber, 2017-02-06:
Thanks. The issue is solved.

We have code coverage right now.
Would it be possible direct gcov to add branch coverage ?
So far we have no been able to.

br,

ST

tgingold, 2017-02-06:
Did you try with gcov -b ?

stdefeber, 2017-02-10:
We are using lcov to get the coverage.
Adding "--rc lcov_branch_coverage=1" helped in getting branch coverage.

We now have to interpret the results ;)

tgingold, 2017-02-15:
Great!

closed 2017-02-15 by tgingold