ghdl/ghdl #268: VCD missing top $scope

markNZed, 2017-02-02:
Our design creates a VCD with a $var declared immediately after the header e.g

```
$timescale
  1 fs
$end
$var reg 1 ! rst $end
```

There should be a $scope declaration before $var ?

tgingold, 2017-02-06:
Yes, I think you're right.  Is it a real problem (i.e. do you have tools that crashes because of this ?)

markNZed, 2017-02-06:
Not a tool but a PLI application from http://chris.spear.net/pli/ we are using to read the VCD and drive inputs to Icarus for Verilog. It is open source so we were able to modify Chris' code as a work around. I don't have access to the VCD standard and Chris had some code for dealing with a "phantom" top level scope. So I'm not sure if it is a bug in the PLI code or a bug in GHDL or both.

svenn71, 2017-06-19:
gtkwave does not re-present the signals on toplevel after a submodule has been visited. Think they can be selected for viewing by using the signal search function, so there is a kind of workaround.

commit cec118dc (2018-07-27): vcd: add global scope.

Fix #268
closed 2018-07-27 by tgingold