ghdl/ghdl #269: Why not to return non-zero exit code on testbench executing fails?

tivaliy, 2017-02-02:
It's interesting is it possible to return non-zero exit code if testbench fails or I miss something?

Let's see some of your examples (https://github.com/tgingold/ghdl/blob/master/doc/Starting_with_GHDL.rst#a-full-adder)
Here is the result of executing your code:
```
root@587c04395e74:/tmp/vhdl# nano adder.vhd 
root@587c04395e74:/tmp/vhdl# ghdl -a adder.vhd adder_tb.vhd 
root@587c04395e74:/tmp/vhdl# ghdl -e adder_tb
root@587c04395e74:/tmp/vhdl# ghdl -r adder_tb
adder_tb.vhd:53:7:@8ns:(assertion note): end of test
root@587c04395e74:/tmp/vhdl# echo $?
0
```
Now, let's change your stimulus pattern, put there some wrong values and check the results:
```
root@587c04395e74:/tmp/vhdl# ghdl -a adder.vhd adder_tb.vhd 
root@587c04395e74:/tmp/vhdl# ghdl -e adder_tb
root@587c04395e74:/tmp/vhdl# ghdl -r adder_tb
adder_tb.vhd:48:10:@1ns:(assertion error): bad sum value
adder_tb.vhd:53:7:@8ns:(assertion note): end of test
root@587c04395e74:/tmp/vhdl# echo $?
0
```

tgingold, 2017-02-06:
Try with --assert-level=error
Any assertion with severity >= error will stop simulation with an error exit status.

closed 2017-02-06 by tgingold

tivaliy, 2017-02-06:
Thanks, it works