ghdl/ghdl #270: Dockerizing GHDL as a step towards the libre and portable VHDL design framework

1138-4EB, 2017-02-07:
After a chat with @Paebbels on [gitter](https://gitter.im/ghdl1/Lobby) last night, I open this issue to show the proof of concept of a possible alternative on how GHDL approaches Continuous Integration and Release publication.

I've known [Docker](https://www.docker.com/) for almost three years now, and I have been following the project with interest. I've sporadically run some containers, but neved went far from the examples/tutorials. Mostly because:

- Even though the idea is to sandbox the applications run in the containers, root privileges are required.
- The Windows version was run on VirtualBox, which, at first, didn't seem very different from actually using a VM.
-Most of the references I found where about web apps, which I'm not interested in. Moreover, it was not recommended for production, so I always left it apart.

Since it's been a while now, the project has improved. When I tried it againt last week, I was really pleased to know that the Windows version can run in the PowerShell (and in MSYS2 too). It seems not to be discouraged for production any more

Inmediately GHDL came to my mind.

- https://docs.docker.com/compose/production/

1138-4EB, 2017-02-07:
# Question 1

Do you find docker interesting enough to give it a try? I can help to adapt the example to your actual requirements.

# Question 2

@tgingold @Paebbels @LarsAsplund

Is it worth creating a GitHub organization to put GHDL, [PoC](https://github.com/VLSI-EDA/PoC), [OSVVM](https://github.com/OSVVM/OSVVM) and [VUnit](https://github.com/VUnit/vunit) together?

I don't mean moving the main repos there, but having a corresponding `*-docker` version. Then, the code development flow (issues, branches,...) would remain unchanged for the developers of the projects.

Each of them can have their own CI system. However, there is no need for them to have multiple build environments. You choose ubuntu/debian or whatever u prefer. Any other is kept at <organization>/ghdl-docker.

tivaliy, 2017-02-16:
@1138-4EB , please check https://hub.docker.com/r/tivalii/ghdl-jenkins-swarm-slave/ . In my case I've created Jenkins swarm client based on GHDL. Then I use it for creating VHDL code check syntax Job in Jenkins. The last one is used to verify VHDL code syntax for Gerrit+Jenkins like CI-system. In future I'm planning to integrate VUNIT to create another Jenkin Job that will run unittests.

closed 2017-02-24 by tgingold

Paebbels, 2017-12-09:
As I relabeld this discussion with the new `Build: Docker` label, I found this maybe related issue: [GHDL with GCC backend as Docker container](../149)