ghdl/ghdl #273: Analysis Crashes on Taking Second-Level Attributes of Array Subtype

preusser, 2017-02-09:
The `'subtype` attribute [LRM §16.2.2, VHDL 2008] gives access to the fully constrained subtype of an object like a variable. This feature works for scalars and arrays. For scalars, it is currently the only way to retrieve information about the exact bounds of the backing subtype by the use of second-level attributes like `'left`. For arrays, such a two-stage is unnecessary but legal. While ghdl manages to retrieve an array subtype as for typing another variable, its analysis chokes when this subtype is queried for a second-level attribute as demonstrated by the testbench below.

```vhdl
entity test is
end test;

architecture tb of test is
begin
  process
    variable a  : integer range 7 downto 0;
    variable b  : a'subtype;
    variable av : bit_vector(7 downto 0);
    variable bv : av'subtype;
  begin
    report integer'image(b'subtype'left);
    report integer'image(bv'left);

    -- The following line crashes the analysis.
    report integer'image(av'subtype'left);

    wait; -- forever
  end process;
end tb;
```

Paebbels, 2017-02-09:
Possible duplicate of [Attribute 'subtype does not work](https://github.com/tgingold/ghdl/issues/253).

preusser, 2017-02-10:
How can a concrete _persisting_ bug be a _duplicate_ of an already resolved issue?

tgingold, 2017-02-24:
I can reproduce the crash.  Investigating.

commit a1567ab5 (2017-02-26): Eval_Array_Attribute: Handle 'Subtype.

Fix #273
closed 2017-02-26 by tgingold