ghdl/ghdl #278: bad type conversion caused crash

react66, 2017-02-12:
I removed the offending statement which was not supposed to be there anyways, but
I have seen a couple crashes due to alias bitfields. It reports the problem but then I
see the ghdl bug spew.

ghdl: compilation error
../rtl=>ghdl -a ram_lut.vhd
../rtl=>ghdl -a ram_lut.vhd
ram_lut.vhd:194:28: can't match type conversion with type array type "std_logic_vector"
ram_lut.vhd:194:28: (location of type conversion)

******************** GHDL Bug occured ****************************
Please report this bug on http://gna.org/projects/ghdl
GHDL release: GHDL 0.33 (20150921) [Dunoon edition]
Compiled with GNAT Version: 4.8
In directory: /home/rob/Desktop/SVN_CLAM/amt1071/ram_lut_tb/rtl/
Command line:
/usr/bin/ghdl1-llvm -P/usr/lib/ghdl/v93/std/ -P/usr/lib/ghdl/v93/ieee/ -c -o ram_lut.o ram_lut.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
Exception name: SYSTEM.ASSERTIONS.ASSERT_FAILURE
Message: no field Type
Call stack traceback locations:
0x7ff17c445901 0x451901 0x5dd90b 0x5bb80a 0x5bb8f3 0x5bd487 0x5bc3e6 0x5bd52d 0x5bd3b1 0x5bd3b1 0x5bd67b 0x5be45a 0x5be4bd 0x5bee79 0x5bf159 0x5f45a3 0x5fc3d3 0x55953e 0x5afbba 0x46eef6 0x433013 0x7ff17b95cf43 0x432450 0xfffffffffffffffe
******************************************************************
ghdl: compilation error

[ram_lut_vhd.txt](https://github.com/tgingold/ghdl/files/769002/ram_lut_vhd.txt)

tgingold, 2017-02-12:
This issue has been fixed since 0.33.  ghdl doesn't crash anymore:

ram_lut.vhdl:194:28:error: can't match type conversion with type array type "std_logic_vector"

closed 2017-02-12 by tgingold

react66, 2017-02-12:
OK, it was only when I wrote bad code anyways,
but I checked and .33 is the current release so you might see more of these until .34 is out.

1138-4EB, 2017-02-12:
@react66 are you working on windows?

react66, 2017-02-12:
No it is ubuntu 14.04.
    
    On 17-02-11 11:15 PM, 1138-4EB wrote:
    
    
      @react66 are you working on windows?
      —
          it on GitHub, or mute
          the thread.

1138-4EB, 2017-02-13:
Then, you can easily compile the latest version yourself, so that the fixes that @tgingold added to the master branch are included in your binary. Just do:

```
$ apt-get update -qq
$ apt-get -y install gnat-4.8 gcc make git zlib1g-dev
$ git clone https://github.com/tgingold/ghdl.git
$ cd ghdl
$ ./configure --prefix=/usr/local
$ make
```

It will be available as 'ghdl_mcode'. You can copy it to `/usr/local` with `make install`. Then, setting the envvar will do:

```
$ make install
$ export GHDL=/usr/local/bin/ghdl
$ $GHDL -v
GHDL 0.34-dev (2016-09-14-358-g87a4f74) [Dunoon edition]
 Compiled with GNAT Version: 4.8
 mcode code generator
Written by Tristan Gingold.
...
```
Yo can test it by running the testsuite:

```
cd testsuite
./testsuite.sh
```

Alternatively, LLVM can be used instead of mcode. The differences are summarized [here](https://github.com/tgingold/ghdl/blob/master/BUILD.txt). To do so, just install a few additional dependencies, and add a couple of arguments to `configure` and `make`:

```
$ apt-get update -qq
$ apt-get -y install gnat-4.8 gcc make git zlib1g-dev llvm-3.5-dev llvm-3.5-tools clang libedit-dev
$ git clone https://github.com/tgingold/ghdl.git
$ cd ghdl
$ ./configure --prefix=/usr/local --with-llvm-config=llvm-config-3.5
$ make CXX=clang++
$ make install
$ export GHDL=/usr/local/bin/ghdl
```