ghdl/ghdl #279: [Question] Which content in `dist` is used on GNU/Linux?

1138-4EB, 2017-02-12:
Is the content in `dist/appveyor`, `dist/gcc` and `dist/mcode` meant for windows only? Nothing is used when compiling on GNU/Linux, isn't it?

Paebbels, 2017-02-12:
The content in `dist/appveyor` is for the AppVeyor flow. AppVeyor is a continuous integration service like Travis-CI, but is offers a Windows Server 2012 R2 machine. The AppVeyor flow has more scripts than the Travis-CI flow, so we put them into a separate directory.

The code in `dist/gcc` and `dist/mcode` is special code needed for the different backends. As you know, GHDL supports three backends:
* **gcc** - GHDL generates an intermediate representation for GCC, which creates an executable. So GHDL acts a a new language frontend like C for the GCC and uses the existing backend to generated e.g. x86(-64) binary code.
* **llvm** - GHDL generates instructions for the LLVM abstract register machine, which then generates x86(-64) instructions for the host system.
* **mcode** - GHDL generates the x86(-64) instructions in memory and executes the model.

mcode is supported on Linux (32/64-bit) and Windows (32-bit).

gcc is currently only supported on Linux, because you need to compile a complete GCC from zero and add GHDL as a frontend into the GCC. Compiling the complete GCC suite plus GHDL takes a lot of time. I think it's not possible to finish the compile task in the bounds of a Travis-CI run.

Does this answer all your questions.

1138-4EB, 2017-02-12:
It does answer why I opened this issue. Thank you.

closed 2017-02-12 by 1138-4EB