ghdl/ghdl #281: Contraint error when running cocotb endian_swapper example

benreynwar, 2017-02-14:
I saw the following bug report when running a cocotb example.
I pulled cocotb today and the hash is 5d6aee25171dcbb45037520554ef5fa5f809418e

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (2016-09-14-366-gde9c5ce) [Dunoon edition]
Compiled with GNAT Version: 4.6
Target: x86_64-pc-linux-gnu
In directory: /home/ben/Code/cocotb/examples/endian_swapper/tests/sim_build/
Command line:
/usr/local/bin/ghdl -r endian_swapper_vhdl --vpi=/home/ben/Code/cocotb/build/libs/x86_64/libvpi.so
Exception CONSTRAINT_ERROR raised
Exception information:
Exception name: CONSTRAINT_ERROR
Message: grt-vpi.adb:1195 discriminant check failed
Call stack traceback locations:
0x482a13 0x482b7b 0x426ac0 0x45cc05 0x45d040 0x45d1a1 0x408c44 0x46f28f 0x629dbc 0x62ef6c 0x62df16 0x4c9141 0x4088ea 0x2b3f85f65f43 0x407ea7 0xfffffffffffffffe
******************************************************************

tgingold, 2017-02-16:
cocotb tests have failed for weeks.  I plan to investigate it sooner or later.

Paebbels, 2017-02-16:
Yes, the Endian Swapper example worked some weeks ago, but on the other hand the Cocotb development is dead since a year! Dozens of issues and pull requests are open.

So I decided to stop my personal investments into Cocotb even if I find it a great idea to test VHDL components. I don't say it was all for nothing - GHDL can now be compiled on Windows with LLVM or mcode backend in MinGW - but all this afford was made to finally run Cocotb on Windows with GHDL.

As I asked half a year ago what's the status and is Cocotb still alive, they answered they will be more active again. I have seen circa 5 merged pull requests from contributors but no Bugfixes.

benreynwar, 2017-02-16:
OK, thanks.  Just doing what the bug report told me to!
It doesn't look with Cocotb is dead, just somewhat time starved.

Paebbels, 2017-02-17:
@benreynwar Please don't get me wrong. I appreciate your bug report to GHDL!

I created an [open discussion at Cocotb](https://github.com/potentialventures/cocotb/issues/513). Last time a wrote a private email and asked for the status and almost nothing happened. This time it's public and some people like my question. I don't want this project to die and I would like to keep the Git forks from diverging. So let's see what happens next :).

P.S. If you want to see an example of a more complex Cocotb testcase see our [sort_lru_list](https://github.com/VLSI-EDA/PoC/blob/master/tb/sort/sort_lru_list_cocotb.py?ts=2) or [cache_par](https://github.com/VLSI-EDA/PoC/blob/master/tb/cache/cache_par_cocotb.py?ts=2) tests.

Martoni, 2017-06-08:
Any clue to solve this bug ?

commit 6d21f84a (2017-06-09): vpi: CbNextSimTime is one-shot.

Fix #281
closed 2017-06-09 by tgingold