ghdl/ghdl #282: Possible to get list of all instances with elaborated generic parameters?

benreynwar, 2017-02-14:
Is it possible to get elaboration information from ghdl?

For example, I have a top level design, which contains many instances of the 'Memory' module but with different 'width' and 'depth' generic parameters.  It would be very useful to be able to get all the (width, depth) pairs that are instantiated so that I could use this information to automate the generation of the compiled memories.

Is this currently possible?  If not, would it be difficult to implement, and what would the best way to go about it be?

Paebbels, 2017-02-14:
You can use cuncurrent assert statements to print the passed parameters.

benreynwar, 2017-02-14:
Ah, great idea!  Thank you!

closed 2017-02-14 by benreynwar