ghdl/ghdl #283: CONSTRAINT_ERROR when using VPI and generate statements with Cocotb

kaolpr, 2017-02-15:
When using VPI interface with Cocotb I'm getting the following error when trying to access submodules internal signals:
```
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 4.9.3
Target: x86_64-linux-gnu
In directory: /home/kaolpr/project_X/sim/sim_build/
Command line:
/home/kaolpr/simulation_framework/lib/ghdl/build/bin//ghdl -r --ieee=synopsys -frelaxed-rules -P/home/kaolpr/simulation_framework/lib/xilinx-vivado/unisim/v08/ --std=08 integration_top --vpi=/home/kaolpr/simulation_framework/lib/cocotb/build/libs/x86_64/libvpi.so --wave=waveform.ghw --ieee-asserts=disable
Exception CONSTRAINT_ERROR raised
Exception information:
Exception name: CONSTRAINT_ERROR
Message: grt-vpi.adb:1397 access check failed
Call stack traceback locations:
0x494413 0x494596 0x494834 0x494b67 0x7f25716cdf63 0x7f25712ac7e3 0x7f25712acb51 0x7f256ec036b4 0x7f2570a4b74f 0x7f2570b7401a 0x7f2570aca2de 0x7f2570a9d1e1 0x7f2570b1131a 0x7f2570a9d1e1 0x7f2570a9d64d 0x7f2570abd806 0x7f2570a4647d 0x7f2570ae4917 0x7f2570a4b7ce 0x7f2570b7401a 0x7f2570a49cfb 0x7f2570b7401a 0x7f2570a49cfb 0x7f2570b7401a 0x7f2570aca2de 0x7f2570a9d1e1 0x7f2570b1131a 0x7f2570a9d1e1 0x7f256ec02499 0x7f25712ab3d4 0x7f25716cf3e7 0x49387d 0x49399d 0x427c0b 0x4676e9 0x467b15 0x467c8d 0x409b7d 0x47bb0d 0x657130 0x65cb9c 0x65b83e 0x4deb1b 0x4097ae 0x7f257190182e 0x4085b7 0xfffffffffffffffe
******************************************************************
```
I did some research and it turns out that the problem occurs at call to `vpi_handle_by_name`.

Sandboxing showed that in presence of generate statements in top architecture referencing to submodules in cocotb testbench leads to CONSTRAINT_ERROR. However, referencing signals and ports of top architecture is just fine.

Now, I'm REALLY not into Ada, but I managed to produce some debugging messages (using Put, I confess...) and it turns out that the problem comes from null reference passed to `Strcasecmp` function (grt-vpi.adb:1385). From what I noticed, when GHDL searches for a name requested by VPI it iterates over names available in a design. It goes fine through names of signals and submodules but when it comes to generate statements (I guess?) a null reference occurs.

Files I used for sandboxing:
**example.py**
```Python
import cocotb
from cocotb.triggers import Timer

@cocotb.test()
def test_transaction(dut):
  
    yield Timer(1)

    dut._log.info("%d" % dut.s_test)
    dut._log.info("%d" % dut.cmp_bar.s_test)
```
**uut.vhd**
```vhdl
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use ieee.numeric_std.all;

entity foo is
  port (
    a : std_logic;
    b : std_logic_vector(7 downto 0)
  );
end entity foo;

architecture RTL of foo is

  signal s_test : std_logic_vector(3 downto 0) := "1111";

begin

  cmp_bar: entity work.bar
    port map(
      a => a
    );
    
  gen_bars: for i in 0 to 1 generate
    cmp_generated_bar: entity work.bar
      port map(
        a => a
      );
  end generate gen_bars;

end architecture;
```
**bar.vhd**
```vhdl
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use ieee.numeric_std.all;

entity Bar is
  port (
    a : std_logic
  );
end entity Bar;

architecture RTL of Bar is

  signal s_test : std_logic_vector(3 downto 0) := "1111";

begin

end architecture;
```
**Modified Makefile.ghdl**
```Makefile
###############################################################################
# Copyright (c) 2014 Potential Ventures Ltd
# All rights reserved.
#
# Redistribution and use in source and binary forms, with or without
# modification, are permitted provided that the following conditions are met:
#     * Redistributions of source code must retain the above copyright
#       notice, this list of conditions and the following disclaimer.
#     * Redistributions in binary form must reproduce the above copyright
#       notice, this list of conditions and the following disclaimer in the
#       documentation and/or other materials provided with the distribution.
#     * Neither the name of Potential Ventures Ltd,
#       SolarFlare Communications Inc nor the
#       names of its contributors may be used to endorse or promote products
#       derived from this software without specific prior written permission.
#
# THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS" AND
# ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED
# WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
# DISCLAIMED. IN NO EVENT SHALL POTENTIAL VENTURES LTD BE LIABLE FOR ANY
# DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES
# (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES;
# LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND
# ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
# (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS
# SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
###############################################################################

ifneq ($(VERILOG_SOURCES),)

results.xml:
	@echo "Skipping simulation as Verilog is not supported on simulator=$(SIM)"
clean::

else

CMD_BIN := ghdl

ifdef GHDL_BIN_DIR
    CMD := $(shell which $(GHDL_BIN_DIR)/$(CMD_BIN) 2>/dev/null)
else
    # auto-detect bin dir from system path
    CMD := $(shell which $(CMD_BIN) 2>/dev/null)
endif

ifeq (, $(CMD))
    $(error "Unable to locate command >$(CMD_BIN)<")
else
	GHDL_BIN_DIR := $(shell dirname $(CMD))
    export GHDL_BIN_DIR
endif

RTL_LIBRARY ?= work

ifdef VHDL_VERSION
	GHDL_STD = --std=$(VHDL_VERSION)
endif

.PHONY: analyse

# Compilation phase
analyse: $(VHDL_SOURCES) $(SIM_BUILD)
	cd $(SIM_BUILD); \
	$(CMD) -i $(GHDL_STD) --ieee=synopsys --mb-comments -fexplicit -frelaxed-rules --work=$(RTL_LIBRARY) -P$(UNISIM_DIR) $(VHDL_SOURCES); \
	$(CMD) -m $(GHDL_STD) $(ELABORATE_ARGS) --ieee=synopsys -frelaxed-rules --work=$(RTL_LIBRARY) -P$(UNISIM_DIR) $(TOPLEVEL)

results.xml: analyse  $(COCOTB_LIBS) $(COCOTB_VPI_LIB)
	cd $(SIM_BUILD); \
	PYTHONPATH=$(PYTHON_SIM_MODULES_PATH):$(LIB_DIR):$(SIM_ROOT):$(PWD):$(PYTHONPATH) LD_LIBRARY_PATH=$(GHDL_BIN_DIR)/../lib:$(LIB_DIR):$(LD_LIBRARY_PATH) MODULE=$(MODULE) \
        TESTCASE=$(TESTCASE) TOPLEVEL=$(TOPLEVEL) TOPLEVEL_LANG=$(TOPLEVEL_LANG) \
	$(CMD) -r --ieee=synopsys -frelaxed-rules -P$(UNISIM_DIR) $(GHDL_STD) $(TOPLEVEL) --vpi=$(LIB_DIR)/libvpi.$(LIB_EXT) $(SIM_ARGS)

clean::
	-@rm -rf $(SIM_BUILD)
endif
```
**Simulation Makefile**
```Makefile
PWD=$(shell pwd)

# Adjust to your needs:
# ======================================================================================================================

# Path to the root folder of the simulation framework
SIM_FRAMEWORK_DIR = /home/kaolpr/simulation_framework

# Top VHDL entity
TOPLEVEL = foo

# Python test file name (without extension)
MODULE = example

# Waveform output file
WAVEFORM_OUTPUT = waveform.ghw

# (Typically) Not to be modified:
# ======================================================================================================================

UNISIM_DIR = $(SIM_FRAMEWORK_DIR)/lib/xilinx-vivado/unisim/v08/
VHDL_VERSION = 08

VHDL_SOURCES = $(shell pwd)/uut.vhd $(shell pwd)/bar.vhd

COCOTB = $(SIM_FRAMEWORK_DIR)/lib/cocotb
GHDL_BIN_DIR = $(SIM_FRAMEWORK_DIR)/lib/ghdl/build/bin/

SIM=ghdl
ANALYSE_ARGS =
ELABORATE_ARGS =
SIM_ARGS = --wave=$(WAVEFORM_OUTPUT) --ieee-asserts=disable

PYTHON_SIM_MODULES_PATH = $(SIM_FRAMEWORK_DIR)/lib/python_modules

include $(COCOTB)/makefiles/Makefile.inc
include $(COCOTB)/makefiles/Makefile.sim
```

tgingold, 2017-06-09:
I can reproduce a crash and I am investigating.

closed 2017-06-09 by tgingold

Martoni, 2017-06-12:
That works !
Thank you very much :)

commit deb8888c (2017-06-09): vpi: avoid a crash on anynomous subtype.

Fix #283